SynthstromAudible/DelugeFirmware #1317: Cracks and pops during playback

Stingmeister, 2024-02-24:
### Please describe the problem:

The problem seems to be only present with specific presets. They are not appearing at the same place so are not suspecting that a file is corrupt. Only 3 tracks playing.

### What is the expected behavior?

Not any cracks and pops during playback...

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_1_0-dev+2024_02_23-b6f9b91.bin

### If possible provide the steps to reproduce the issue and upload additional media:

[SONG015.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14391078/SONG015.zip)

m-m-adams, 2024-02-24:
This is the sound of culling - there's only 3 tracks but with the long releases you end up with 30-40 voices plus the 2 delays and reverb running

That said I don't think this should cull this much, I pushed some more tweaks to #1316 that might help

m-m-adams, 2024-02-25:
Closed by #1316

closed 2024-02-25 by m-m-adams