SynthstromAudible/DelugeFirmware #1355: Freeze (crash) selecting Filters via Sound Menu

MichaelinVic, 2024-02-26:
### Please describe the problem:

Freeze selecting Filters via Sound Menu. Seems to happen only for some songs. I have attached an offending one.

### What is the expected behavior?

Selecting Filters via the Song Menu shows filter options for clip.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

c1.1.0

### If possible provide the steps to reproduce the issue and upload additional media:

1. Load  song
2. spSelect
3. tSelect to Filters
4. spSelect

[SONG2026..xml.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14411978/SONG2026.xml.zip)

sapphire-arches, 2024-03-03:
This occurs specifically with FM filters.

m-m-adams, 2024-03-03:
It's an infinite loop since none of the filter items are relevant. We need to add an isRelevant to the filter item itself

PR #1417 (2024-03-03): Bugfix/menu relevance
Fix #1329 
Fix #1355
closed 2024-03-03 by stellar-aria