SynthstromAudible/DelugeFirmware #1358: OLED seems broken

patriciachadwick62, 2024-02-27:
### Please describe the problem:

With community firmware
[image]
the nightly looks like this too

back to offical release all fine again
[image]

### What is the expected behavior?

seeing the ... on the display

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

sapphire-arches, 2024-02-27:
Hi, thanks for the bug report. A couple of things that could help us narrow down what the issue is:

  - Did you power-cycle after the initial flash? We rely on the bootloader to tell us if we're on OLED or 7seg hardware and sometimes this logic gets confused after the first flash. Subsequent boots should have no problem
  - Is your OLED a retrofit?
  - Do you see a synthstrom logo during the boot process?

If you have a retrofit and do not see a synthstrom logo it's likely whoever retrofit your deluge did not update the bootloader.

patriciachadwick62, 2024-02-27:
Thank Y for the fast response

- I did,  and test after every try
- Yes its a retrofit made in Netherlands
- No don't see the Logo
-

And how can it be done can I do it myself

sapphire-arches, 2024-02-27:
It must be v8 for OLED. You'll need to use official firmware to flash, and double-check to make sure the file is copied to the SD card successfully and is the only .bin on the SD card.

patriciachadwick62, 2024-02-28:
Thank Y
Got it from Synthstrom
so the community firmware is now on

Thank Y

sapphire-arches, 2024-02-28:
Glad to hear it's working for you now! I'll close this issue, let us know if you run in to any other problems.

closed 2024-02-28 by sapphire-arches