SynthstromAudible/DelugeFirmware #1380: Long audio samples can drop out.

AaronRito, 2024-02-29:
### Please describe the problem:

Sometimes when performing with a long audio sample, the sample sound will drop out. Muting and unmuting the clip can resolve but the behavior is unpredictable. 

### What is the expected behavior?

That audio samples never drop.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### If possible provide the steps to reproduce the issue and upload additional media:

Starting in grid view, load a long audio sample into a kit. Use grid view to copy the clip and load another long audio sample into the same kit. When changing rows in grid view, the sample can drop out randomly. 

Can we get play direct from SD card???

AaronRito, 2024-02-29:
You can see the failure in action here. The sample is loaded into column 1 using a kit, that contains other long samples as sounds. As I switch to the verse, you can hear the sample (music) drop out. 
https://youtu.be/VfkaAEkg-yw?t=2262

m-m-adams, 2024-02-29:
If you overload the cpu voices are stopped to recover. Generally the longest running voices will be stopped, but you have a couple options to control it:
1. Use audio clips - they're more lightweight and won't be stopped
2. Set the priority to high for the voice you don't want to stop. High priority voices will only be stopped if there are no medium or low priority voices sounding

closed 2024-03-02 by m-m-adams

AaronRito, 2024-03-02:
@m-m-adams is there any way to know what capacity the cpu has left?

sapphire-arches, 2024-03-02:
generally only by noticing culling, unfortunately. The rendering process is highly dynamic and can be limited by several distinct factors (DSP capacity and SD bandwidth are the 2 big ones) and it is difficult to predict which will limit performance from within the current software architecture.