SynthstromAudible/DelugeFirmware #1407: Crash when entering filters menu item on FM synth

sichtbeton, 2024-03-02:
### Please describe the problem:

When you select the „filter „ menu item and press select, the deluge crashes. This happens in a blank new FM synth clip. There is no QR code, it just stops working and needs a reboot.

[image]

### What is the expected behavior?

No crash and actually enter the menu.

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.2.0-NIGHTLY-24D683

### If possible provide the steps to reproduce the issue and upload additional media:

New synth clip
Change to FM mode
Enter sound menu via push “select”
Find “filter”
Push select

joxihan, 2024-03-02:
isn't it a duplicate of https://github.com/SynthstromAudible/DelugeFirmware/issues/1355 ?

sapphire-arches, 2024-03-03:
Yep, dupe of 1355.

closed 2024-03-03 by sapphire-arches