SynthstromAudible/DelugeFirmware #1408: Collect media not working

sichtbeton, 2024-03-02:
### Please describe the problem:

When trying to save a song with files, the following pop up comes up. Song is not saved.

https://github.com/SynthstromAudible/DelugeFirmware/assets/62035517/e3acded2-8679-4e6d-bdd9-b00d723c97e9

### What is the expected behavior?

Save song and no pop up

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.2.0-NIGHTLY-24D683

### If possible provide the steps to reproduce the issue and upload additional media:

Press save long
Select “collect media”
Press save again
Bug

m-m-adams, 2024-03-02:
https://github.com/SynthstromAudible/DelugeFirmware/assets/52469396/ae5adc5e-8a50-4dbe-ac87-b4fa42927983

Can't reproduce - will need more info in the issue

sichtbeton, 2024-03-04:
Strange, now I also cant reproduce.

closed 2024-03-04 by sichtbeton