SynthstromAudible/DelugeFirmware #1422: Midi Notes lengths get truncated when recording in arranger mode - still not fixed

Radius9999, 2024-03-04:
### Please describe the problem:

When recording a midi track (to play an external synth) in arranger mode, long notes get truncated to almost 0 length intermittently.  Only occurs with quantization on.  When quantization is off, notes record properly.  (This does not happen when in song mode.)

### What is the expected behavior?

Note offs should not be quantized when quantization is enabled in arranger mode.  Note lengths should remain as played.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

C1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

https://github.com/SynthstromAudible/DelugeFirmware/assets/147354026/01157925-d96c-44e2-ae15-76efa1cf9b87

m-m-adams, 2024-03-04:
What are your quantization settings?

Radius9999, 2024-03-04:
It doesn't matter what the quantization settings are.  I've tried different settings.  But it's the note lengths we're talking about.  Those should not be quantized at all.

m-m-adams, 2024-03-04:
Can you please share your quantization settings? The bug is clearly related to quantization, and I have not been able to reproduce it since limiting the quantization menu to the range of values actually supported by the deluge

m-m-adams, 2024-03-04:
You can also try clearing your settings (hold select while booting) to see if that fixes the issue

Radius9999, 2024-03-04:
I've already tried clearing settings.  By my quantization settings, you mean ....  I have it set to 16th notes under Settings-Recording-Quantization.  Is there some other place to adjust quantization?

m-m-adams, 2024-03-04:
Nope that's it. I'll try 16th notes, I have mine at 64ths and haven't been noticing issues but possibly it's just close enough that I don't notice

m-m-adams, 2024-03-04:
Ok yeah, reproduced

PR #1691 (2024-04-09): Fix note-on ignoring during quantized record
Rather than using a boolean that gets out of sync easily, we record the actual relative time stamp at which we should start paying attention.

There's probably gremlins in the reverse clip recording code, but I don't think this makes it any worse at least.

Fixes #1422.
closed 2024-04-09 by m-m-adams