SynthstromAudible/DelugeFirmware #1425: Cannot record midi when deluge is sending out midi CC data at the same time

saynotohugs, 2024-03-04:
### Please describe the problem:

Is this expected behavior?

### What is the expected behavior?

I'd love if you could send out CC data to external gear and record at the same time

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus, Nightly

### What is the firmware name:

1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

Turn on
Create midi track 1, channel one. Set 1 bar of automation data, loop 1 bar length.

Create midi track 8, channel 8. Arm track for midi recording with learn button. 

in clip view of track 8, hit record and then play. 

After recording, the length of automation track is now 8 and no midi data has been recorded on channel 8. 

If I delete track 1 I can record on track 8 without problems.

m-m-adams, 2024-03-04:
You have to learn the track to the midi input - check the manual for details, but it's learn+audition pads and then send a note or CC

saynotohugs, 2024-03-04:
Hey m-m-adams,

"Create midi track 8, channel 8. Arm track for midi recording with learn button."

I did that

saynotohugs, 2024-03-04:
When doing the same thing - but using an external keyboard instead of OP-Z, the automation track has the length changed even though it's not paired/midi learned.
But the midi recording itself is without problems.

Using the same channel on both the (outgoing) automation track and on the midi track on which I record.  

Using another channel on the (outgoing) automation track than the midi track on which I record - same behaviour.

saynotohugs, 2024-03-04:
have a look

saynotohugs, 2024-03-04:
Uploading test.mp4…

m-m-adams, 2024-03-16:
Can you take a video? I think I'm having trouble following the description

m-m-adams, 2024-04-27:
Closing for now, but if you can take a video or provide more info I'll look into it again

closed 2024-04-27 by m-m-adams