SynthstromAudible/DelugeFirmware #1431: Midi Notes lengths get truncated to 128th when recording on a synth track or converting from midi track to synth track

saynotohugs, 2024-03-04:
### Please describe the problem:

With quantization OFF:

Is it expected behaviour, that when recording midi with external keyboard to the midi in all note values are truncated to 128ths? 

Is it expected behaviouror that converting from midi track to synth track will truncate all note values to 128ths? 

### What is the expected behavior?

I'd love to keep the note lenghts as played?

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

With quantization OFF:

1: Open deluge
2: midi learn on synth track with external keyboard via midi in
3: record something
all events have the same length (128ths)

1: Open deluge
2: midi learn on midi track with external keyboard via midi in
3: record something
4: convert to synth track
all events are truncated to the same length (128ths)

sapphire-arches, 2024-03-05:
This probably has the same root cause as #1422

sapphire-arches, 2024-04-09:
This should be fixed in tomorrows nightly/beta builds, please reopen if this problem reproduces still.

closed 2024-04-09 by sapphire-arches