SynthstromAudible/DelugeFirmware #1447: Sample RAM not ideal. Worse than with official Deluge FW.

Zedehzedeh, 2024-03-06:
### Please describe the problem:

• Longer Loop samples got cut in half or not played at all. This is completely differend with the latest official Deluge FW.
• When using an external sequencer (masterclock) to play Deluge patches in addition to the internal Deluge sequencer, there is a clicking sound at the start of the pattern (see video).

https://github.com/SynthstromAudible/DelugeFirmware/assets/162515840/c5150598-6b4b-428e-b035-72ce5e50aefb

### What is the expected behavior?

Is it possible to fix the RAM problem. There where never more than 3.5 MB of samples loaded and the longer ones got cut off or not played at all. There have been a lot of effects and CC-data controlling almost everything via Novation Launchcontrol + an external Sequencer (torso T1). Now I am working with the synthstrom 4.1.4 FW and I do not have sample issues at all.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

1.0.1 Amadeus

### If possible provide the steps to reproduce the issue and upload additional media:

Clicking Sound on 1st step of the sequence.
samples cut off or not played at all

sapphire-arches, 2024-03-07:
To diagnose this issue, it would be helpful to have the "collect-all" song file. In general we should perform better than official 4.1.4, but our culling choices can be different.

Setting the priority on voices can help work around this issue.

sapphire-arches, 2024-03-07:
oh, there's also significant performance improvements in the latest 1.1 beta over 1.0. Please try the latest beta (after backing up your SD card) and see if the issue reproduces.

Zedehzedeh, 2024-03-07:
I am sorry – it is not possible to have a "collect-all" Song file since I changed some of the samples drastically. I am sorry ...
I will try the 1.1 and I will let you know if I encounter some of the issues mentioned above.

Zedehzedeh, 2024-03-07:
OK, I tried the 1.1 and it is not satisfying. There are no more issues with samples that get cut off/not play at all. But there is an issue with clicking sounds of samples. Attached you will find the collected data of the song. You may have to play with the master volumes of the tracks since I use a midi controller to define the volume and I do not know the volume saved in that song. The last track is controlled by an external sequencer. As soon as I use that sequencer (master clock) the cracking sounds get really bad, especially on the first step of the pattern.

For comparison I added 2 videos. Identical settings different FW:
1) FW 4.1.4 clean sound – noch cracking noises
2) FW 1.1 cracking gets really bad as soon as I deploy the external sequencer

Attachements: 2x Video + Collected Song-Data

https://github.com/SynthstromAudible/DelugeFirmware/assets/162515840/cedb95cd-31a9-4ee8-94ea-2cfe1c201e82

https://github.com/SynthstromAudible/DelugeFirmware/assets/162515840/3a9fceb4-c4d9-4a64-adb4-998bc10a150f

[collect.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14523910/collect.zip)

sapphire-arches, 2024-03-07:
Thanks, hopefully I or another developer have some time soon to take a look at this.

m-m-adams, 2024-03-07:
The bug without external midi is from a bad interaction between monophonic synths and the cull system - the change in voice is done as a 'fast release' but with a slower release rate than a soft cull. The 'definitelyCull' logic changes soft culls to hard culls if the voice to be released has already been put in fast release, which in this instance causes the mono synth note changes to become hard culls. We probably need to set a flag for 'been culled' or check the release rate as well in that logic

m-m-adams, 2024-03-07:
https://github.com/SynthstromAudible/DelugeFirmware/assets/52469396/95ac2bff-95de-426c-aea7-9a634767c522

Checking against official and this song culls and clicks very often there as well.

Official also gets 8 fewer overall voices.

Zedehzedeh, 2024-03-07:
Yeppp – same effect....
Actually, I don't underrstand a word, I am not a software engeneer and english is not my motherlanguage – so mr. Adams, your post gives me a hart time. Is there anything I can do to make results better or is that a matter of the FW? Sorry ...

Zedehzedeh, 2024-03-07:
Ahhh thanks so much – that I can understand. And thanks a lot for the advice concerning the clicks in the middle of the patterns. And thanks for your wonderful work on the FW.

PR #1452 (2024-03-09): fix force culling resulting in hard culls with monophonic voices
Adds an immediate off function to voices - this is similar to a hard cull but with one final render pass. Since volume changes are interpolated and since this is only called when the number of samples in the window is high it reduces the clicking

Adds a check for whether a fast release is actually fast - if not it sets a new fast release rate. Fixes an issue where mono synths and choked samples would be in the 'fast release' state, but releasing 15x slower than a fast release from a cull and get accidentally skipped.

Fix #1447
PR #1455 (2024-03-09): fix force culling resulting in hard culls with monophonic voices
Adds an immediate off function to voices - this is similar to a hard cull but with one final render pass. Since volume changes are interpolated and since this is only called when the number of samples in the window is high it reduces the clicking

Adds a check for whether a fast release is actually fast - if not it sets a new fast release rate. Fixes an issue where mono synths and choked samples would be in the 'fast release' state, but releasing 15x slower than a fast release from a cull and get accidentally skipped.

Fix #1447
closed 2024-03-09 by sapphire-arches