SynthstromAudible/DelugeFirmware #1454: Launching a clip while stuttering makes most buttons unresponsive

chee, 2024-03-08:
### Please describe the problem:

in 🔵 **SONG** mode, in the ROWS or PERFORMANCE view, launching a clip while you are stuttering makes most of the buttons stop doing anything. (you can still use 🟢 **PLAY**, 🔵 **TRIPLETS VIEW**, ⚫ **SELECT** and the 🟩 **LAUNCH** pads)

### What is the expected behavior?

i'd expect to still be able to use the deluge normally!

### Is there a relevant Pull request?

no!!

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

d3c519361e4e08ba92265c67a6f2b7dcf4dc92af

### If possible provide the steps to reproduce the issue and upload additional media:

you'll need a song with at least 2 rows

- enter 🔵 **SONG** mode (ROWS view)
- 🟩 arm at least 1 of the rows
  - leave at least 1 row 🟥 unarmed
- press 🟢 **PLAY**
- press 🟠 **STUTTER/CUSTOM 1**
- press and hold the top gold performance knob
- while held, 🟩 launch the unarmed clip
- release the top gold performance knob
- note how you can now no longer use most of the pads or buttons! 

launching another clip once you've released the knob will return you to a normal working state

chee, 2024-03-08:
this is in c1.0.1 too 👀 .

also, it happens whether `STUTTER RATE QUANTIZE` community feature is off or on

chee, 2024-03-08:
it does not happen on 4.1.4

sichtbeton, 2024-03-22:
Can reproduce. Very odd one indeed.

closed 2024-03-23 by m-m-adams