SynthstromAudible/DelugeFirmware #1466: Midi preset names don't allow you to see the midi channel setup for that midi track

soymonitus, 2024-03-10:
### Please describe the problem:

When you do Save + Midi in a midi track and give it a name, it will show:
```
           MIDI CHANNEL
MY AWESOME MIDI PRESET
```
When you use the select knob to change the midi channel you can only see that it changes because MIDI CHANNEL changes to MIDI ZONE after 16 increments of the Select knob, and then goes to INTERNAL, and then back to MIDI CHANNEL. It is changing the midi channel behind the scenes, but the name of the preset doesn't allow you to see it

### What is the expected behavior?

when changing the midi channel, the midi preset name show hide and instead show the midi channel number, and after you stop scrolling it goes back to showing the preset name

### Is there a relevant Pull request?

Nightly

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

soymonitus, 2024-03-10:
@m-m-adams

m-m-adams, 2024-03-10:
I agree but I'm unlikely to find time to change it myself - the current behaviour is just a result of how names interact with midi channels

alifeinbinary, 2024-03-20:
@soymonitus there's a lot more to build out in the MIDI Presets feature. This Save+MIDI, Load+MIDI is just a part of the  early work, which got pulled to the community repo. If @m-m-adams is too busy at the moment to develop it further then we should try to find other contributors. 

Here are the following relevant threads on the community Discord, which discuss how MIDI presets might be implemented. 
[https://discord.com/channels/608916579421257728/1211653958662234184](https://discord.com/channels/608916579421257728/1211653958662234184)
[https://discord.com/channels/608916579421257728/1212935088401285121](https://discord.com/channels/608916579421257728/1212935088401285121)

I develop apps on Mac and don't have the jlink tool for debugging but I'm becoming more and more motivated to get set up with on in order to contribute, especially to this feature.

m-m-adams, 2024-03-20:
Yeah, names existing at all is just a happy accident of the fact that when deluge reads a clip from file it sets the filename as the synth name. I've only done the file storage side of this so far, I haven't looked at midi presets at all beyond the necessary hooks to allow the current information to be saved and loaded

closed 2024-03-22 by m-m-adams