SynthstromAudible/DelugeFirmware #1482: Row of leds start blinking in time with the mastercompressor

x3003, 2024-03-15:
### Please describe the problem:

When applying the mastercompressor, the row of synth, kit, midi & cv also start blinking in time with the mastercompressor's lights.
These are very faint, hard to see, but they'll light up.

### What is the expected behavior?

The leds from the kit/synth/midi/cv not blinking.

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_03_14-6f984b2

m-m-adams, 2024-03-15:
Not a bug, it's the light from the compressor LED shining through the other buttons

closed 2024-03-15 by m-m-adams

x3003, 2024-03-15:
eh, why ? you can already see the leds next to the gold knob go up and down....

m-m-adams, 2024-03-16:
It's a hardware flaw - literally the light from the gold knob LED showing in the other pads. There's nothing to be done about it

You could maybe open your deluge up and stick some tape around the LEDs?