SynthstromAudible/DelugeFirmware #1483: Mater volume in song mode not returned to full level

supchikov, 2024-03-15:
### Please describe the problem:

Mater volume in song mode not returned to full level

Steps to reproduce bug
create some track with some sounds with overall volume about 80% or something 
Switch to song mode
Reduce master volume to 0
Try to set volume level back to normal
Result:
Volume not returning to full level before it been set to nil, only about 50% from original.

### What is the expected behavior?

Mater volume in song mode not returned to full level

Steps to reproduce bug
create some track with some sounds with overall volume about 80% or something 
Switch to song mode
Reduce master volume to 0
Try to set volume level back to normal
Result:
Volume not returning to full level before it been set to nil, only about 50% from original.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

Last beta

### If possible provide the steps to reproduce the issue and upload additional media:

Mater volume in song mode not returned to full level

Steps to reproduce bug
create some track with some sounds with overall volume about 80% or something 
Switch to song mode
Reduce master volume to 0
Try to set volume level back to normal
Result:
Volume not returning to full level before it been set to nil, only about 50% from original.

m-m-adams, 2024-03-15:
Confirmed, works fine until the volume has touched zero. After touching zero it seems to be halfed in value so that volume 50 is equivalent to original volume 25

ok-reza, 2024-03-20:
i've mentioned this on the discord, but so i don't forget:
im finding that most loaded songs are too loud now, to the point where many songs are clipping on load.
for most cases, if i bring song level to 0 and back to original level value, the song level seems to be as it should with no clipping and at a level i've come to expect.

however there has been at least one song i've found that when touching zero and back, the volume of the song sounds twice as quiet as it should be.

once i get a chance i'll upload songs that can replicate both instances of this with.

betonizm, 2024-03-29:
After the fix, pan doesn't work properly.  when crossing zero, the panorama fixes the last value and is no longer edited

m-m-adams, 2024-03-31:
Also once volume has gone to zero and then turned back up pan stops working

m-m-adams, 2024-04-03:
Fixed by #1635

closed 2024-04-03 by m-m-adams