SynthstromAudible/DelugeFirmware #1496: E404 - On synth clip switch from arpeggiator OFF to ON

foolofaperegrin, 2024-03-17:
### Please describe the problem:

Two synth clips of the same patch in a song - Clip A has the arpeggiator OFF, Clip B has the arpeggiator ON.
The patch is mapped to a MIDI keyboard.

When a MIDI keyboard note is held while Clip A is playing, triggering Clip B causes the deluge to crash when it launches.

It won't crash if the note is briefly released for the transition, nor will it happen if Clip B is switched to Clip A (i.e. in the opposite order, Arp ON to OFF).

### What is the expected behavior?

We'd expect the deluge to not crash and simply launch the clip with the arpeggiator on. I suspect it may be to do with the MIDI note on/off being lost with a held note during the switch but I'm not sure.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

deluge-c1_0_1

### If possible provide the steps to reproduce the issue and upload additional media:

Make a new song, make two clips of synth patch 0. 

Set second clip Arpeggiator to "UP" (or any ON state)
Map synth patch 0 to an external MIDI keyboard.

Play the first clip. 
Hold a MIDI note on the keyboard.
Trigger Second Clip and keep the keyboard note held.

Deluge should crash with E404 on display.

soymonitus, 2024-03-21:
I tested it in current beta & nightly and it doesn't crash. Could you try any of those?

soymonitus, 2024-03-21:
I installed 1.0.1 and could reproduce it. So definitely this is already fixed in 1.1 and can be closed

seangoodvibes, 2024-03-22:
Closing per @soymonitus tests above. Thanks for confirming!

closed 2024-03-22 by seangoodvibes