SynthstromAudible/DelugeFirmware #1609: Song does not save arp sync setting

colinmeyer, 2024-03-27:
### Please describe the problem:

Change arp sync setting (either by clip -> arp -> sync or by sync shortcut key in arp column), save song, reload song, and arp sync is back to default 16th notes

### What is the expected behavior?

Song should retain the arp sync setting, e.g. Off

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

1b376355

### If possible provide the steps to reproduce the issue and upload additional media:

1. new song
2. select button (clip menu) -> arp -> sync -> off
3. save song
4. load same song
5. select button (clip menu) -> arp -> sync (now shows 16th instead of off)

sapphire-arches, 2024-03-29:
I think this matches official behavior -- if the arpegiator itself is off, we don't save any of the arpegiator state, and the code for that check goes back to the initial OS release. I'm not opposed to changing this, but for now I'm going to remove it from the 1.1 milestone.

colinmeyer, 2024-03-29:
Rate and Gate get saved, regardless of Arp on or off. Sync and Octaves do not get saved

sapphire-arches, 2024-03-29:
can you upload a song file with a preset that doesn't save correctly? Also, what do you have `DEFAULTS > RESOLUTION` set to?

m-m-adams, 2024-03-31:
Rate and gate are internal automatable parameters in a synth that get passed to the arp, so they get saved differently - this is unintuitive but consistent with official

soymonitus, 2024-03-31:
Would it be bad to save it regardless of arp off/on state? @m-m-adams 
The code change seems just to remove the if condition here (and some other places more)

[image]

closed 2024-04-01 by sapphire-arches