SynthstromAudible/DelugeFirmware #1638: Volume dips when using certain functions on audio tracks.

Gibbonnaboa, 2024-04-01:
### Please describe the problem:

I recorded an audio clip from stereo in then experimented with automating parameters.
 I first tried EQ and also tried the filters, I always got a considerable volume dip as soon as I moved a gold knob or started to draw in any kind of automation using the pads.
 Deleting the automation using back and top black knob wouldn't return the volume to previous level, only using the gold knobs to return the parameters setting to where it was originally would increase the volume again on the filters, I couldn't fix it with the eq. 
 Volume also dips in the same way when I use the new reverb.

### What is the expected behavior?

Only the parameter being automated just change, volume shouldn't be affected by changes to filter or EQ directly. 
 Volume should stay the same when reverb is turned on. 

### Is there a relevant Pull request?

Not sure

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.20-nightly+2024_03_31-32aaeac

### If possible provide the steps to reproduce the issue and upload additional media:

1.record audio clip
2.open automation view
3.select filter LPF via pad
4.turn gold knob or draw automation
5.volume dips wherever parameter value is not at default level. 

1.record audio clip
2.turn on reverb
3.volume dips.

sapphire-arches, 2024-04-01:
This should be fixed in the next nightly.

Gibbonnaboa, 2024-04-01:
Prompt, I like it.

Gibbonnaboa, 2024-04-01:
I've tested this and it's still not quite right, it's much better but there's still a noticeable dip in volume.

sapphire-arches, 2024-04-01:
what's the exact version reported in the settings > version menu?

Gibbonnaboa, 2024-04-01:
On the 7 segment display it says 1.20-nightly-7aa29db.
 It's the latest BIN file I found, I downloaded it about 2 hours a go, deluge-v1_2_0-nightly+2024_04_01-7aa29db

sapphire-arches, 2024-04-01:
That incorporates all the fixes we've built so far, I think. There must still be some gain problem.

Gibbonnaboa, 2024-04-01:
Okay cool, I'll keep an eye on it with the nightly updates, automation on audio tracks is quite the boon for me.

sapphire-arches, 2024-04-02:
I have a potential fix at #1655.

@Gibbonnaboa, if you have some time to spend running tests for me I'd appreciate it if you could grab the PR build (see [this wiki page](https://github.com/SynthstromAudible/DelugeFirmware/wiki#navigating-github) if you need help navigating the github UI to find it, or drop by the Synthstrom Discord and I can help you interactively). To test, you'll need to create an audio clip with a filter enabled.

That change increases the base makeup gain in the compressor to the limit of what we can support without getting clipping inside the compressor. The compressor could be reworked to add further internal headroom, at the potential cost of some sample fidelity.

sapphire-arches, 2024-04-02:
To be clear, there's still a volume change when the compressor is enabled, but hopefully it's now gone from "jarring" to "sorta acceptable".

seangoodvibes, 2024-04-02:
@Gibbonnaboa here's the link to download it if you can't find it: https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/8516810611?pr=1655

Gibbonnaboa, 2024-04-02:
Thanks, I'll have a look into it when I get some time. I'm pretty busy for the next few days I'm afraid, i may get chance to have a quick look but I'd like to try and help. 

I'm not following why the compressor is now involved here?

m-m-adams, 2024-04-02:
It shouldn't be. It's not supposed to render if its threshold is at zero

closed 2024-04-03 by m-m-adams