SynthstromAudible/DelugeFirmware #1646: Been getting clicks and pops pretty bad on both community 1.0.1 and the latest nightly builds

mattyharm, 2024-04-01:
### Please describe the problem:

I've been getting audio clicks and pops on my OLED deluge a lot on the community FW - it seems like it's an issue with samples specifically, like if i have one or two sample-based instruments as well as drums triggering.  Sidechaining may or may not make it worse - sometimes it seems like it does, and other times I get them regardless.  I used to run into this issue (as well as dropouts) on the official firmware if I was running a ton of tracks with a lot going on, which I just chalked up to CPU limitations, but this will happen on the community firmware (for me at least) with like, one or two instruments and a drum track and no intensive FX or anything.

In case anyone has time to mess with this or confirm whether you're also getting this behavior - I've included a simple song file that I quickly mocked up that uses two Boards of Deluge 6 sounds (sample-based), one factory 808 drum track, and one audio track that I recorded in from an external synth.  On both 1.0.1 and today's nightly build, if I hit play, I get random clicks and pops, but if I roll back to 4.1.4 OLED, it's smooth.

Thanks - appreciate y'all!

song file with collected media here - https://drive.google.com/drive/folders/1HIOzE7uf9E7YXGJQVgReU9ngSPtvSR5M?usp=sharing

### What is the expected behavior?

No clicks or pops, especially with only 3-4 tracks running

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus, Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_04_01-7aa29db.bin

### If possible provide the steps to reproduce the issue and upload additional media:

Load song file and song folder both into the "SONGS" folder on your sd card
Turn on Deluge
Load song (I collected the media so you should have no trouble with samples/etc)
Hit play and let it run through while listening for random pops and clicks

download here - https://drive.google.com/drive/folders/1HIOzE7uf9E7YXGJQVgReU9ngSPtvSR5M?usp=sharing

sapphire-arches, 2024-04-01:
On nightly and beta we indicate hard culling by flashing the "play" LED. If you see that happening the Deluge is generally unhappy, so you'll want to resample some stuff or otherwise find ways of reducing the number of active voices.

mattyharm, 2024-04-01:
I really appreciate the quick response @sapphire-arches!  That makes sense, and also answers another question I had (that blinking play light - I noticed that the other day and thought my unit was glitching out).  But I will say, I'm getting pops and clicks way more often than the light blinks.

For now I'll just try to heavily limit my use of sample based voices / use resampling, but do you think this is something that will get better as the community releases continue?  Like I said before, if I roll back to 4.1.4, I don't have these problems unless I'm running wayyyy more tracks, but I love a lot of the features that the community releases have added.

Thanks,
Matt

sapphire-arches, 2024-04-02:
Yes, we're constantly working to improve performance. Our steady state performance (no new note generation) performance is better than official in most cases, we just make different culling and sample streaming decisions that can make our culls more noticable in some scenarios.

sapphire-arches, 2024-04-02:
Yes, we're constantly working to improve performance. Our steady state performance (no new note generation) performance is better than official in most cases, we just make different culling and sample streaming decisions that can make our culls more noticable in some scenarios.

Having songs like the one you provided really help us know exactly how people are pushing the device, so thanks for that!

m-m-adams, 2024-04-02:
How were you comparing the song on each firmware? Attempting to open "click example 4.xml" reliably crashes my deluge running official firmware

seangoodvibes, 2024-04-02:
> I really appreciate the quick response @sapphire-arches! That makes sense, and also answers another question I had (that blinking play light - I noticed that the other day and thought my unit was [...]

fyi we noticed there was a bug in the play button blinking. it should've been blocking wayyyy more. so we've now fixed that. so I expect you'll see it blinking any time to get pops or clicks now.

mattyharm, 2024-04-02:
> How were you comparing the song on each firmware? Attempting to open "click example 4.xml" reliably crashes my deluge running official firmware

Thanks for the responses!  I just tried that using the file I uploaded and got the same result - I had created the song on 4.1.4 and then bounced back and forth several times between different firmware versions and got the result reliably that there were clicks and pops on community but not 4.1.4, but then when I collected media/saved the version I uploaded for y'all, I was on the latest nightly version, and that must have created a backwards compatibility issue.  Next time I get a chance, I'll make another similar track from 4.1.4 and see if I can get that same behavior of it clicking on the nightly but not 4.1.4, and upload a version that was saved out of 4.1.4.

Crsko, 2024-04-04:
You may also be having an issue similar to what I get when the audio input is left on with monitoring, where, without a source providing input noise, (as in, nothing is playing into the input,) I receive pops/ clicks.

mattyharm, 2024-04-04:
Hi all @m-m-adams @sapphire-arches @seangoodvibes - I made a new quick song project that I feel embodies the issue I've been having with random clicks/pops pretty well.  I didn't use BOD sounds - there's just one factory synth that I made into an 1/8th note arpeggiator, one bass sound that I made (but it shouldn't be particularly CPU-hungry, I don't think), and one factory drum kit.  Seems like it really shouldn't be hitting a limit, but I'm getting clicks at random throughout.  

I've included a link to download the project file (which was created on nightly build deluge-v1_2_0-nightly+2024_04_02-45de91f.bin) below, as well as a link to a quick recording I made of the issue happening at the beginning of the song in case you can't reproduce.  If you load the file and go right to the arrangement view and hit play, I'm finding that most of the time, I start getting random clicks when the bass synth comes in after 8 measures, and then I hear them at random throughout.  Oh, and, none of the sounds have that click inherent in them - like, for instance, when the bass sound comes in in the video I made, it might sound like it's a phase re-triggering issue or something, but that sound never clicks on its own.

Would really appreciate any insight you have!

Download song: https://drive.google.com/drive/folders/1Fr07KVwVxBVgithTFC-cqqU80sUJ4nE_?usp=sharing
Example video: https://photos.app.goo.gl/RumoChsE5CdDGMMH6

m-m-adams, 2024-04-04:
Great thanks! I'll take a look in the next few days

MichaelinVic, 2024-04-12:
I'm getting the blinking play button and lots of pops and crackles on just one track the new BOD07 - 62 - Some Kind of Harpth.

seangoodvibes, 2024-04-12:
> I'm getting the blinking play button and lots of pops and crackles on just one track the new BOD07 - 62 - Some Kind of Harpth. On tonights nightly.
> 
> 
> 
> Edit testing others in BOD7 collection and it seems its not isolated to just that one. 
> 
> 
> 
> Edit 2: Actually its pretty much happening all the time even a few basic synth only tracks...

All of those BoD7 presets are really heavy. They are mostly sample based, a lot are multi samples, lots of wave table based synths also.

They also have long release times which further compounds the issue.

The play button blinking rapidly means that they are eating up all the deluge's cpu and voices/samples are now being culled back - hence the pops and clicks.

Basically you need to make those presets lighter and / or resample them in order to use them.

I'm not sure what we can do to make them useable.

MichaelinVic, 2024-04-12:
> All of those BoD7 presets are really heavy. They are mostly sample based, a lot are multi samples, lots of wave table based synths also.
>
> They also have long release times which further compounds [...]

Fair. I've downgraded to c1.01 and I couldn't load the previous song because the Deluge would crash. However it was simple enough to rebuild and the pops and clicks are barely present with the same tracks/notes. It seems the current nightly is reducing cpu overhead vs c1.01.

MichaelinVic, 2024-04-12:
Further testing this morning on the the latest official V4.1.4 - for OLED and I can run several clips with no pops or cracks present at all (that I can hear) I've tested up to 5 BoD7 as well as a synth kit clip.

mattyharm, 2024-04-12:
> Further testing this morning on the the latest official V4.1.4 - for OLED and I can run several clips with no pops or cracks present at all (that I can hear) I've tested up to 5 BoD7 as well as a synth kit clip running simultaneously.

Thanks for adding to this @MichaelinVic - I've just accepted at this point that the BOD sounds that involve samples are pretty CPU heavy and might be a "only one or two per song" type of thing, but this was my experience too a few weeks ago when I started this thread - 4.1.4 in particular handling them better than the current official community release and the nightlies.  Any chance you could share that latest test song (with media collected) saved from 4.1.4 so that it can be opened both in 4.1.4 and the latest nightly?  I'm just curious to see if I can replicate your experience with your exact song in case it helps the developers zero in on this.  I'm also gonna do my own new test as well

mattyharm, 2024-04-12:
@MichaelinVic Wait though, how did you get the BoD7 patches to work on 4.1.4?  I just tried and they all sound super garbled and wrong, I assume because they were created for the community release

MichaelinVic, 2024-04-12:
> @MichaelinVic Wait though, how did you get the BoD7 patches to work on 4.1.4? I just tried and they all sound super garbled and wrong, I assume because they were created for the community release

Really? Not on my unit. I did have to recreate the "song" since the one created on the latest lightly would just crash on attempt to load on 4.1.4. I can for sure upload anything but I'm not sure how helpful it will be its not really a song also :) just a bunch of clips with notes to test the capability.

mattyharm, 2024-04-12:
> > @MichaelinVic Wait though, how did you get the BoD7 patches to work on 4.1.4? I just tried and they all sound super garbled and wrong, I assume because they were created for the community release
> 
> Really? Not on my unit. I did have to recreate the "song" since the one created on the latest lightly would just crash on attempt to load on 4.1.4 (as expected). I can for sure upload anything but I'm not sure how helpful it will be its not really a song also :) just a bunch of clips with notes to test the capability.

@MichaelinVic that's ok, haha my test songs have been not "songs" also.  i would be pretty curious if you wouldn't mind - more than anything now because it's interesting that your BOD7 patches seem to be working on 4.1.4.  I'm curious if I load your song with media collected whether my unit will tolerate those patches

MichaelinVic, 2024-04-12:
> @MichaelinVic that's ok, haha my test songs have been not "songs" also. i would be pretty curious if you wouldn't mind - more than anything now because it's interesting that your BOD7 patches seem [...]

Here ya go: [https://drive.google.com/drive/folders/1U3FIdJ7nBsikZR2uhwRWzG4zvuFyjSmJ?usp=share_link](url)

MichaelinVic, 2024-04-18:
Created a 12 bar loop with a synth kit and 7 BOD7 tracks. This plays fine on ym Deluge on 4.1.4 with the rare glitch. This is pushing it I'm sure and in reality I'd never have this many synth sounds in a single "song" but I'm just looking for the edges:

https://drive.google.com/drive/folders/1G2wa4FhPvW0j-KKAqpGlVoaqPIDEp4wf?usp=sharing

m-m-adams, 2024-04-19:
Trying to check these out but I don't have access - I sent a request for it

mattyharm, 2024-04-19:
@m-m-adams @MichaelinVic Ah, I missed that you had granted me access to the first song you shared a few days ago - thanks for that!  A couple observations from playing with it for a few minutes just now -

1) I'm absolutely having that same experience - the song clicks and pops like crazy on the latest nightly build, but doesn't at all on 4.1.4 OLED
2) BUT, a lot of the sounds in BOD7 sound really garbled and different when you load them on 4.1.4 - I assume because they make use of non-official-firmware community features.  It's a little hard to tell having to go back and forth between my computer and the deluge to switch firmware back and forth whether any of these particular sounds you chose are in that boat, but I wonder if part of the issue is that they just aren't using as much CPU on 4.1.4 because they aren't actually operating at full potential or something?  Curious what some of the developers here think.  

Also, @MichaelinVic if you can replicate that experience of some of the BOD7 sounds sounding different depending on whether you load them in 4.1.4 or community - one that's really obvious on my unit is the first sound, 01-Aural.  It sounds nice and clear (albeit with pitch warble) on community and all garbled and kind of low-sample-rate-ish on 4.1.4 on my unit.

m-m-adams, 2024-04-19:
If you could try out the build in #1743 and let me know what you think that would be great. It just sets the threshold for culling to match 1.0.1 instead of having an extra middle level

m-m-adams, 2024-04-19:
@mattyharm I'd tend to agree on the BOD issue - I know they're big on the S+H and random walk LFO shapes, which could both be confused for unintended clicking. I don't own them though so it's tough to troubleshoot

mattyharm, 2024-04-19:
> @mattyharm I'd tend to agree on the BOD issue - I know they're big on the S+H and random walk LFO shapes, which could both be confused for unintended clicking. I don't own them though so it's tough to troubleshoot

@m-m-adams I'd be happy to share them if you want to try messing with them - would even be happy to kick another $12 to BOD to keep it above board if you thought having access to all 75 sounds could help in what y'all are doing.  But in the meantime, @MichaelinVic did collect all the media for that first song he shared, and you can definitely hear the clicking going crazy on community firmware in a way that I really don't think is just the s&h lfo, from having played with that feature a lot

mattyharm, 2024-04-19:
> If you could try out the build in #1743 and let me know what you think that would be great. It just sets the threshold for culling to match 1.0.1 instead of having an extra middle level

I'd be happy to - but can you point me towards where that firmware file is?  I'm not seeing a download link anywhere there

m-m-adams, 2024-04-19:
> @m-m-adams I'd be happy to share them if you want to try messing with them - would even be happy to kick another $12 to BOD to keep it above board if you thought having access to all 75 sounds could [...]

oh agreed - I didn't realize they were BOD presets in that song

m-m-adams, 2024-04-19:
> I'd be happy to - but can you point me towards where that firmware file is? I'm not seeing a download link anywhere there

https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/8758662972?pr=1743

Download the release build bundle from there, it's got a .bin file in it that you can install. It's definitely better but I don't know if it's a total fix, I haven't compared to official

mattyharm, 2024-04-19:
> > > If you could try out the build in #1743 and let me know what you think that would be great. It just sets the threshold for culling to match 1.0.1 instead of having an extra middle level
> > 
> > 
> > I'd be happy to - but can you point me towards where that firmware file is? I'm not seeing a download link anywhere there
> 
> https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/8759694864
> 
> Download the release build bundle from there and it's got a .bin file in it that you can install. It's definitely better but I don't know if it's a total fix, I haven't compared to official

Thanks @m-m-adams - just tried it, and it's definitely better than the previous community nightly I tried @MichaelinVic's song on, but I'd say it's still having more audible clicks.  Sort of subjective, but the clicks sound different to me - in the previous nightly build version, they were frequent loud super-short clicks, and in this build bundle you just linked me to, they're quieter, less frequent, and sound kinda more like short white noise impulses or something.  In any case, fwiw, I wasn't hearing even those types of clicks on 4.1.4 on his sample song, but I know that's a tough comparison due to the fact that we're talking about BOD7 sounds

MichaelinVic, 2024-04-19:
> @m-m-adams @MichaelinVic Ah, I missed that you had granted me access to the first song you shared a few days ago - thanks for that! A couple observations from playing with it for a few minutes just [...]

I can in a bit. FWIW here is an audio clip of it playing in 4.1.4 is this what 01-Aural sounds like to you in 4.1.4? 
[REC00041.txt](https://github.com/SynthstromAudible/DelugeFirmware/files/15046035/REC00041.txt)

mattyharm, 2024-04-19:
> > @m-m-adams @MichaelinVic Ah, I missed that you had granted me access to the first song you shared a few days ago - thanks for that! A couple observations from playing with it for a few minutes just now -
> > 
> > 1. I'm absolutely having that same experience - the song clicks and pops like crazy on the latest nightly build, but doesn't at all on 4.1.4 OLED
> > 2. BUT, a lot of the sounds in BOD7 sound really garbled and different when you load them on 4.1.4 - I assume because they make use of non-official-firmware community features.  It's a little hard to tell having to go back and forth between my computer and the deluge to switch firmware back and forth whether any of these particular sounds you chose are in that boat, but I wonder if part of the issue is that they just aren't using as much CPU on 4.1.4 because they aren't actually operating at full potential or something?  Curious what some of the developers here think.
> > 
> > Also, @MichaelinVic if you can replicate that experience of some of the BOD7 sounds sounding different depending on whether you load them in 4.1.4 or community - one that's really obvious on my unit is the first sound, 01-Aural. It sounds nice and clear (albeit with pitch warble) on community and all garbled and kind of low-sample-rate-ish on 4.1.4 on my unit.
> 
> I can in a bit. FWIW here is an audio clip of it playing in 4.1.4 is this what 01-Aural sounds like to you in 4.1.4? Change .txt to .WAV [REC00041.txt](https://github.com/SynthstromAudible/DelugeFirmware/files/15046035/REC00041.txt)

@MichaelinVic Thanks - that's what it sounds like on mine too in 4.1.4 but it's garbled compared to what it sounds like on community - that version sounds like there's some amount of sample rate degradation or something to it, whereas if you load it on community/nightlies it doesn't have that aspect at all.  Some of the BOD7 sounds are way worse than that when you load them on 4.1.4, too.  This all makes sense, since he made BOD7 to make use of community features, and like Mark pointed out somewhere, if nothing else, a lot of them have the s&h/random LFO shapes baked into them

MichaelinVic, 2024-04-19:
> @MichaelinVic Thanks - that's what it sounds like on mine too in 4.1.4 but it's garbled compared to what it sounds like on community - that version sounds like there's some amount of sample rate [...]

Ya that makes sense. So I guess some or all of BOD7 is for community firmware and when used there can only be used minimally like granular effects... This is going to start getting confusing :)

m-m-adams, 2024-04-20:
New build up here with some more changes - https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/8761020867

MichaelinVic, 2024-04-20:
> New build up here with some more changes - https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/8761020867

Testing: Test Bod7 4.1.4.XML

I get play blinking on the first clip BOD7-30 but seems to be playing normally

So seems to be handling the first 3 clips:
1. BOD7-30
2. BOD7-38
3. BOD7-71

I get noticeable clips/chirps when starting play on the 4th clip BOD7-69

I notice when I stop play on 4 that the clips/chirps continue despite the clip no longer playing.

Definitely improved, nice work!

MichaelinVic, 2024-04-20:
> @MichaelinVic Thanks - that's what it sounds like on mine too in 4.1.4 but it's garbled compared to what it sounds like on community - that version sounds like there's some amount of sample rate [...]

I can hear a slight difference in BOD7-01-Aural not massive but there is some additional clarity when playing on community.

mattyharm, 2024-04-20:
> Testing: Test Bod7 4.1.4.XML

Yeah, maybe that one wasn't the best example, but I think what you're hearing is the 4.1.4 version not knowing what to do with the random LFO (and maybe other new features) that was added in community.  There are other BOD7 patches that just sound like total garbage when played on 4.1.4.  Anyways, thanks for confirming - I just wanted to make sure something wasn't wrong with my unit

PR #1743 (2024-04-21): Bugfix - move force cull to old hard cull threshold
Force cull is a less clicky hard cull. I'd placed it in between soft and hard culling, however there doesn't seem to be a downside to just doing it instead of hard culling. Worst case scenario the high load lasts through one more routine call which isn't really that bad, especially since if the render itself is slow we'll redo it immediately afterwards

Fix #1646
PR #1746 (2024-04-21): Bugfix - move force cull to old hard cull threshold
Force cull is a less clicky hard cull. I'd placed it in between soft and hard culling, however there doesn't seem to be a downside to just doing it instead of hard culling. Worst case scenario the high load lasts through one more routine call which isn't really that bad, especially since if the render itself is slow we'll redo it immediately afterwards

Fix #1646
closed 2024-04-21 by seangoodvibes