SynthstromAudible/DelugeFirmware #1673: Midi CC on Audio Track crashes Deluge

scnroy, 2024-04-05:
### Please describe the problem:

After loading a previously saved song, when I adjust an encoder on my Faderfox EC4 that's mapped to a few params on an audio track, the Deluge crashes.

HPF, and HPF Resonance are fine, strangely. (Channel 1, CC 72/73)

Master Level (Channel 1, CC 76)
[image]

Reverb (Channel 1, CC 68)
[image]

Delay (Channel 1, CC 64)
[image]

Delay Rate (Channel 1, CC 65)
[image]

### What is the expected behavior?

Deluge does not crash

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.1.0 beta

### If possible provide the steps to reproduce the issue and upload additional media:

Here's what I saw:

1. turn on deluge
2. load song (attached)
3. spin encoder
4. 💥

To reproduce

1. create a song
2. add some stuff
3. create an audio track
4. map some external midi controller to master level
5. save song
6. turn off deluge
7. turn on deluge
8. load song
9. spin encoder
10. 💥?

(I just tried to do this and it seems like my learned midi params aren't saving to the Audio Track? Works fine until i switch to a new song and back. Maybe another issue?)

[0 42.XML.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14890285/0.42.XML.zip)

soymonitus, 2024-04-12:
[image]

0x200d73df: AudioOutput::setActiveClip(ModelStackWithTimelineCounter*, PgmChangeSend) at :?
0x200d73b9: AudioOutput::setActiveClip(ModelStackWithTimelineCounter*, PgmChangeSend) at /src/deluge/processing/audio_output.cpp:395
0x200af193: InstrumentClipMinder::makeCurrentClipActiveOnInstrumentIfPossible(ModelStack*) at /src/deluge/model/clip/instrument_clip_minder.cpp:558
0x200a56b3: getCurrentClip() at /src/deluge/model/song/song.cpp:65
 (inlined by) getSelectedClip(ModelStack*) at /src/deluge/io/midi/midi_follow.cpp:159

same just having Midi follow feedback enabled and just converting a synth clip to audio clip in Row mode.

closed 2024-04-13 by sapphire-arches

scnroy, 2024-04-20:
Still seeing this on the release from 4/17 (6e5412d)

Was the fix included in that build?