SynthstromAudible/DelugeFirmware #1681: MIDI Loopback can create infinite loops

sapphire-arches, 2024-04-07:
### Please describe the problem:

When MIDI loopback is enabled, learning a channel to itself and playing a note results in an infinite loop leading to a crash.

### What is the expected behavior?

We don't try to generate infinite note ons.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

c642a4d470610a5b2cc21243dcd1f8d0a1ab72f7

### If possible provide the steps to reproduce the issue and upload additional media:

- Switch initial track to MIDI
- Go to song view and enable MIDI loopback (selpress -> MIDI loopback -> enabled)
- Re-enter MIDI track
- hold LEARN+audition pad in the midi track
- Send a note-on via DIN MIDI.

m-m-adams, 2024-04-27:
Maybe we want to ignore incoming midi notes if they're on the loopback device and have the same channel as the midi clip's output?

sapphire-arches, 2024-04-27:
Can we easily determine if a new mapping will create a closed loop and reject it? That or put a reasonable (3 or 4) limit on the note generation recursion.

m-m-adams, 2024-04-27:
I can't think of a way to do it directly - loopback is at the end of the actual midi engine output, so it doesn't have the source clip. We could refactor to pass it along but at that point we should probably restructure to pass messages directly between clips without faking midi anyway. 

In the clip midi message received function I think we can do something like `ignore if (device == loopbackDevice && inputchannel == outputchannel)` since we know that would make a loop? Not sure if there's a downside

sapphire-arches, 2024-04-27:
that doesn't help if the loop has an intermediate clip
 - in: 1, out: 2
 - in: 2, out: 3
 - in: 3, out: 1

m-m-adams, 2024-04-28:
Ah yeah. I think that might fall under "don't do this"

sapphire-arches, 2024-04-28:
we could do a DFS over midi clips but I'm not sure if we can do that in a way that isn't like quadratic in the number of clips until we have direct event delivery rather than midi offering

m-m-adams, 2024-04-28:
Oh it could go on the learning function - if loopback is enabled then keep an adjacency list of MIDI clips inputs/outputs and block changes that would make a loop?

sapphire-arches, 2024-04-28:
can clips have different learnings within a track (i.e. is the learn on the output or on the clip?) if it's on the clip that might not notice loops created by currently muted clip.

m-m-adams, 2024-04-28:
I dont think so, the midi channel and learned midi is held by the output instrument

commit 7dc9e3cb (2024-05-12): midi: avoid infinite recursion when loopback is enabled (#1891)

Fixes #1681
PR #1891 (2024-05-12): midi: avoid infinite recursion when loopback is enabled
Fixes #1681
commit 3b83c5f3 (2024-05-12): midi: avoid infinite recursion when loopback is enabled (#1891)

Fixes #1681

midi engine: fix 1.1 build
closed 2024-05-12 by sapphire-arches