SynthstromAudible/DelugeFirmware #1697: master clip volume does not hold

worker-gray, 2024-04-09:
### Please describe the problem:

I have a clip that keeps returning to a previously set volume despite numerous attempts to delete volume automation and set manually.
I deleted the volume automation the regular way (shift/press gold knob) and in automation view.
I manually set the master clip volume to 25, and it jumps right back up to 35, even after saving.

### What is the expected behavior?

set volume stays static without automation

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

Amadeus 1.0

### If possible provide the steps to reproduce the issue and upload additional media:

Hit 'play' and volume jumps back to previously set level. Hold shift and press 'master volume' gold knob to delete automation (if any) and manually set volume to 25.
Switch to automation view, select master volume for entire clip-- hold shift and push gold knob to delete automation (if any) then manually set volume.

closed 2024-04-09 by worker-gray

sapphire-arches, 2024-04-09:
It would be helpful to have the song file that exhibits problematic behavior, since it seems like that clip is in a very particular state. Also, it would be good to try the 1.1 beta build to see if the problem has already been fixed.

worker-gray, 2024-04-09:
I turned it off/on after trying to record a video and tried it again-- it worked.