SynthstromAudible/DelugeFirmware #1702: Incorrect 7SEG abbreviations in "menu_hierarchies.md

ok-reza, 2024-04-11:
### Please describe the problem:

In 'Audio Clip Menu' enter section 'Audio Source'.

Disabled should be (OFF) not (DISA)
All monitored inputs should add a period to 7seg abbreviations _eg. left input monitoring will be (LEFT.)_
(BAL) should be changed to (BALA)
Deluge Mix (pre FX) should be (MIX)
Deluge Output (post FX) should be (OUTP)

As far as I can tell, these abbreviations are displayed correctly in a recently new 1.2 nightly on an OLED display. But there seems to be a bug where you can enter this menu and get the OLED text entries displayed while still emulating the 7SEG display (eg. both monitored and unmonitored BALANCED entries appear as BAL. on the emulated 7seg display because it's receiving OLED text entries and OLED abbreviates balance to BAL. INPUT)

### What is the expected behavior?

Consistent 7SEG abreviations in menu_hierarchies

Bug fix so you can't receive OLED entries while emulating 7SEG display.

### Is there a relevant Pull request?

https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/menu_hierarchies.md

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

ok-reza, 2024-04-11:
I found a way to reproduce incorrect emulation behavior (make sure 7SEG display is set to TOGGLE in community settings first):

1) Turn on deluge
2) Enter Song View
3) Convert first clip to an Audio Clip
4) Enter Audio Clip Input settings (LEARN/INPUT + press row of audio clip in Song View)
5) While OLED audio clip Input settings are displayed, toggle 7SEG display shortcut (SHIFT+LEARN/INPUT+AFFECT ENTIRE)
6) Note that the display doesn't change immediately, turn the SELECT knob once and it will switch to the 7SEG display.
7) Note that 7SEG abbreviations are ignored, similarly to how the 7seg abbreviations were originally written in menu_hierarchies.md

If you enter 7seg display before entering audio input (restart deluge to do so), then the proper 7SEG abbreviations display.

seangoodvibes, 2024-04-12:
Partially fixed by https://github.com/SynthstromAudible/DelugeFirmware/pull/1701

m-m-adams, 2024-08-08:
Functionally this is fixed, the 7seg emulation is mainly for dev testing so I don't know that the remaining issue is a big deal

closed 2024-08-08 by m-m-adams