SynthstromAudible/DelugeFirmware #1707: Midi takeover modes "pickup" and "scale" are reacting only on slow CC changes

therealstaubi, 2024-04-11:
### Please describe the problem:

Midi takeover modes "pickup" and "scale" are reacting only on slow CC changes. When turning a USB-Midi controller faster than under two seconds from CC value 127 down to 0 midi takeover mode cannot keep up, hangs or displays values like -793. 
Sometimes I observed that the Deluge even does not react to external MIDI CC anymore. 
Happens in both "pickup" and "scale" mode.

Hint: I checked the midi controller CC output (intech studio GRID PO16) with a MIDI monitor. It sends proper values from 0 to 127.

### What is the expected behavior?

Midi takeover modes "pickup" and "scale" follow external USB-MIDI in real time like in "Jump" mode.

### Is there a relevant Pull request?

I did not find any

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

deluge-v1_2_0-nightly+2024_04_11-60ad680.bin

### If possible provide the steps to reproduce the issue and upload additional media:

- create new song
- witch Midi takeover mode to "pickup" or "scale"
- add synth
- go to clip view and add some notes
- use MIDI follow mode to control current active synth voice
- attach external MIDI controller and turn knob to turn e.g. reverb (CC91) up and down
- observe above behaviour

Used Midi controller: intech studio GRID PO16 connected via USB

soymonitus, 2024-04-12:
In order for Takeover to work, the increments in CC value can't be higher than +-5.

`constexpr int32_t kMIDITakeoverKnobSyncThreshold = 5;`

This limit could be increased but it cannot be super high as it would defeat the purpose of PICKUP mode. The idea for the user is to do a slow swipe to  "catch" the current value and then they can use the knob with faster movements.

seangoodvibes, 2024-04-12:
> Understood. I do not know the exact implementation. Couldn't this be solved by some code like
> 
> if (CCval_controller_t-1 <= CCval_Deluge) // Case1: old controller value is coming from low end { if (CCval_controller_current >= CCval_Deluge) // has new incoming controller value reached Deluge value? { CCval_Deluge = CCval_controller_current; // behave like in jump mode; } else // Case2: controller value is coming from high end { if (CCval_controller_current <= CCval_Deluge) // has new incoming controller value reached Deluge value? { CCval_Deluge = CCval_controller_current; // behave like in jump mode; } } }

Thx for that, yea that's the kind of logic I'm looking to implement to address this

I did implement code that stores the previous controller's value in order to be able to detect the direction that the knob is being turned and I do compare the current value received to the deluge value.

What I think I'm missing is the "reached deluge value" aspect so that it doesn't jump.

I'll hopefully get this fixed soon.

I have an electra one controller with really fast knobs so I'm able to re-produce the problem you reported.

therealstaubi, 2024-04-12:
Great that you are able to reproduce the problem. Electra one is a cool controller. The intech studio GRID controllers also can execute LUA code. Very handy sometimes...

I am happy that my limited Arduino skills are sufficient to contribute to the community firmware. 
Please raise your hand if you need a tester for the new code.

therealstaubi, 2024-04-18:
Addition: the crashes are not caused by the takeover modes. 
They seem to be a general issue between the intech studio grid controllers and the Deluges USB-host implementation. Proof: I connect the intech studio grid controllers via an external USB MIDI host and DIN MIDI to the Deluge.
Result: the controller connection is stable, no crashes occur, Deluge reacts fast to CC changes. Nevertheless the above described takeover behaviour is still there.

sapphire-arches, 2024-04-21:
Let's use #1748 to track the crash and this bug for the MIDI pickup bug.

nikodemus, 2024-05-01:
I came here to report the same takeover bug with Intech Grid controllers using current beta (downloaded on May 1st). No special code is running (out of the box config). Looking at the incoming values with MidiBerry when connected to the computer, a fast slider push generates eg:

0, 1, 5, 16, 32, 79, 100, 115, 124, 127

I think this has more to do with the relatively short slider than anything else, maybe?

Both pickup and scale methods were broken by this, so had to use jump.

Some thoughts:

Could we assume only one external controller per CC, and worry about takeover only va Deluge's own controls? In that case:

- For each mapped control, remember the last CC and the current value (set by CC or internal controller).

- If the new CC matches the current value, we're done.

- If the old CC and the current value match, new CC is valid even if it jumps: there have been no other changes it could be confused with. This fixes all cases of "jumpy controls" like these, no matter how much they jump.

- If the old CC and the current value don't match, we have the following caes:

1. Current value is between old CC and new CC. Consider this a completed takeover: the new CC is valid. This could happen eg. by slightly adjusting via Deluge gold encoder, and then moving a jumpy slider very fast.

2. New CC is between old CC and current value. Value is invalid. Either consider this a scaled change, inferring the direction from the old CC, or use pickup logic and drop it.

3. Old CC is between new CC and current valid. Value is invalid. Either consider this a scaled change, inferring the direction from the old CC, or use pickup logic and drop it.

Does this make sense, or am I awake too late? 😅

seangoodvibes, 2024-05-02:
> I came here to report the same takeover bug with Intech Grid controllers using current beta (downloaded on May 1st). No special code is running (out of the box config). Looking at the incoming values with MidiBerry when connected to the computer, a fast slider push generates eg:
> 
> 0, 1, 5, 16, 32, 79, 100, 115, 124, 127
> 
> I think this has more to do with the relatively short slider than anything else, maybe?
> 
> Both pickup and scale methods were broken by this, so had to use jump.
> 
> Some thoughts:
> 
> Could we assume only one external controller per CC, and worry about takeover only va Deluge's own controls? In that case:
> 
> * For each mapped control, remember the last CC and the current value (set by CC or internal controller).
> * If the new CC matches the current value, we're done.
> * If the old CC and the current value match, new CC is valid even if it jumps: there have been no other changes it could be confused with. This fixes all cases of "jumpy controls" like these, no matter how much they jump.
> * If the old CC and the current value don't match, we have the following caes:
> 
> 1. Current value is between old CC and new CC. Consider this a completed takeover: the new CC is valid. This could happen eg. by slightly adjusting via Deluge gold encoder, and then moving a jumpy slider very fast.
> 2. New CC is between old CC and current value. Value is invalid. Either consider this a scaled change, inferring the direction from the old CC, or use pickup logic and drop it.
> 3. Old CC is between new CC and current valid. Value is invalid. Either consider this a scaled change, inferring the direction from the old CC, or use pickup logic and drop it.
> 
> Does this make sense, or am I awake too late? 😅
> 
> EDIT: read upthread more carefully, and realized pretty much this exact solution had been discussed already. My bad, sorry for the noise.

Very valid points and thoughts, thanks for sharing!

I've actually gotten around to fixing this.

So far I've gotten picked up fixed and still working on getting value scaling working.

My pickup fix is exactly as has been suggested - it assumes a single controller, stores the previous cc value and using the previous cc value and current cc value it infers a direction that your controllers knob is turning (left or right, up or down) and then it compares that to the deluge's current value to determine if it has met or surpassed the deluge knob value in the direction that the controller cc is turning and then it operates like jump.

Value scaling is a bit more tricky but I'm confident I'll get it.

Cheers and thank you for your patience!

seangoodvibes, 2024-05-04:
@therealstaubi + @nikodemus 

I have a fix for the pickup and value scaling issues here if you want to try it out!: https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/8954062472?pr=1831

closed 2024-05-06 by m-m-adams