SynthstromAudible/DelugeFirmware #1709: Loops/samples loaded into audio tracks no longer automatically stretching to closest tempo-synched length

mattyharm, 2024-04-12:
### Please describe the problem:

Just noticed today that at some point fairly recently (not sure exactly which nightly build - I've been downloading them every few days - but I'm almost positive this feature was still working last week), the functionality seems to have broken where when you load a loop into an audio track, the deluge automatically time-stretches or -shrinks it to fit the closest obvious number of measures length to match your tempo.

So for instance, I have a bunch of shaker / tambo / hihat loops organized and titled by BPM that I'll sometimes reach for, and up until today, they would always lock to tempo, so like, if I'm at 96BPM and I load a 90BPM 2-bar shaker loop into an empty audio clip, the deluge would speed it up slightly to match 2 bars at 94BPM.  Today on deluge-v1_2_0-nightly+2024_04_11-60ad680.bin, I tried that a bunch of times and it was just loading the samples in at their original length and speed.  Not a huge deal because it's easy enough to quickly adjust the clip length and force it into tempo, but that was a nice workflow feature if it can be rescued!

### What is the expected behavior?

Deluge locks audio loops loaded into empty audio clips to the closest 1/2/4/8/etc measure length to match the tempo

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_04_11-60ad680.bin

seangoodvibes, 2024-04-12:
Fixed!

closed 2024-04-12 by seangoodvibes

mattyharm, 2024-04-12:
> Fixed! See #1712

Thanks!