SynthstromAudible/DelugeFirmware #1717: Crash E412 while browsing presets (after loading song)

bnhrsch, 2024-04-13:
### Please describe the problem:

The Deluge will crash while browsing synth presets as soon as you reach a Preset that is already in use in a muted clip. This only works if the song has been **saved and loaded at least once** 

### What is the expected behavior?

No crash

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.2.0.NIGHTLY-C6C2DF3

### If possible provide the steps to reproduce the issue and upload additional media:

1. Create new song with 2 Synth-Clips
2. Save Song
**3. Load Song**
4. Mute Synth Clip 1
5. Go into Synth Clip 2 and Scroll to the same Preset as Clip 1

The Deluge will crash with E412 as soon as you reach the already in use Synth patch.

sapphire-arches, 2024-04-21:
I'm not able to reproduce this with a basic song file using the factory Biolab and Rich Saw Bass presets. Can you upload the song file that crashes?

m-m-adams, 2024-04-21:
Should be fixed by #1747

closed 2024-04-21 by m-m-adams