SynthstromAudible/DelugeFirmware #1718: Incorrect label for LPF Drive parameter when mapped to a custom control

neilalbrock, 2024-04-13:
### Please describe the problem:

UI label shows: "LFP MORPH:   <value>"

### What is the expected behavior?

UI label should show: "LFP DRIVE:   <value>"

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_04_13-c6c2df3

### If possible provide the steps to reproduce the issue and upload additional media:

Use the standard procedure to map the LPF Drive parameter to Custom 1, 2 or 3 and then change the value using the encoder. The display shows: "LPF MORPH:  <value>"

neilalbrock, 2024-04-14:
Just reviewing other issues and wondering if this is related to #1629

sapphire-arches, 2024-04-14:
yeah it's likely the same underlying problem, and would be good to fix both together.

seangoodvibes, 2024-04-22:
> yeah it's likely the same underlying problem, and would be good to fix both together.

in this it's the Param::getParamDisplayName function which can only return one string for each param kind and param ID combination at the moment. Logic would need to be added that would say "if param = morph, check filter type and return string accordingly"

neilalbrock, 2024-04-23:
Appreciate the feedback @sapphire-arches and @seangoodvibes

Looking at the way the filters are configured and then looking the the code you mentioned, I can see the issue. I guess it would be logical to check filter mode and use that to switch between strings. I can see that LPF could be "Morph" or "Drive" and the HPF could be "Morph" or "FM", depending on which mode they are in.