SynthstromAudible/DelugeFirmware #1725: Sticky shift window is (subjectively) too short

dkrieger, 2024-04-13:
### Please describe the problem:

Idk if this was introduced between 1.0.1 stable and 1.1 4/12 nightly, or if I'm just noticing now by coincidence, but I find I have to be unreasonably (IMO) fast in releasing shift if I want it to stick.

### What is the expected behavior?

Shift pressed and released within say ~200ms ought to stay on

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_1_0-beta+2024_04_12-4979336bin

### If possible provide the steps to reproduce the issue and upload additional media:

1. Ensure sticky shift is enabled
2. Experiment with press/release durations
3. See if you agree the window is too short (or conversely if a slightly longer window would reasonably be mistaken for an attempt to _not_ engage sticky shift

dkrieger, 2024-04-13:
Consider that if another button is pressed within a longer window, it could (one would think) cancel the sticky shift toggle, mitigating the likelihood of accidentally engaging sticky shift toggle

ok-reza, 2024-04-17:
from my recent further review of enabling sticky shift, i'm still able to accidentally enable it while either working on a creating a song or when live performing with my deluge (i live arrange via mute/unmuting clips, including holding shift while doing so for instant mute/unmuting). so in my case, the sticky shift window is still too short as it disrupts my normal use with the deluge. however, perhaps you're suggestion would be good for solving my issue since a quick instant mute/unmute requires a quick shift press plus another button.

but i think instead of trying to find a way to accommodate one short press of shift to toggle sticky shift on or off, a better shortcut should replace it. one that is a) still one handed b) doesn't rely on a quick press (good for accessibility) and c) isn't simply one button that is widely used with many deluge operations

possible alternate toggles:
1. learn/input + sync scale + shift (3 button combo)
2. hold shift + press learn/input two times (2 button combo with a double tap)
3. hold learn/input + press shift three times (2 button combo with a triple tap)

m-m-adams, 2024-04-19:
I'm strongly against an alternate shortcut. It's hard enough to remember all of them as is. Additionally a key goal of the feature is one handed usability and your suggestions would make that more difficult. I can definitely not manage any of those while my focus is on my other hand 

Making the window longer will make your problem worse @ok-reza - I'm not sure why you say it's too short, if anything what your describing is the window being too long

dkrieger, 2024-04-20:
@m-m-adams does shift revert to momentary behavior if there is another button pressed even if shift is released before the sticky shift deadline lapses? If not, I think that (coupled with a customizable deadline duration) should make most people happy most of the time.

Also seconding that I would rather cope with current state than have an alternate shortcut. This kind of ui is notoriously difficult and subjective to dial to everyone's taste, and all things considered I think it's currently perfectly reasonable if less than optimal (again, in my subjective opinion)

sapphire-arches, 2024-04-20:
yeah the intention is for shift to not stick if another button is pressed before shift is released. It's possible that isn't working for some shortcuts, if you can find a pad which doesn't work that way we can fix it

sapphire-arches, 2024-04-20:
yeah the intention is for shift to not stick if another button is pressed before shift is released. It's possible that isn't working for some shortcuts, if you can find a pad which doesn't work that way we can fix it.

panicmechanic, 2024-04-21:
What about a sub-menu in the main menu settings that allows the user to pick from a predefined range of timing windows?

I'd also like a slightly longer window.

sapphire-arches, 2024-04-21:
yes, configurable short press time is something we can (and probably should) implement.

ok-reza, 2024-04-24:
> I'm strongly against an alternate shortcut. It's hard enough to remember all of them as is. Additionally a key goal of the feature is one handed usability and your suggestions would make that more [...]

yes, my fault, made my post pretty late at night. yes the window is still too long in my case, and it seems to be quite short for some people to successfully toggle as it is. having a configurable short press time really just sounds like a bandaid to a non-ideal shortcut. for one, it will allow users to set a window long enough to easily be enabled during normal use.

my suggestions were all with one handed usability in mind actually. yes, ideally it's not three buttons, but if there's a non zero percent chance that someone can enable sticky shift without intending to, this it's just plain old bad UX. there's a reason why you can't enabled sticky shift on a keyboard for a computer by pressing shift very quickly.

although it seems like multi-taps and holds were explored in the past, maybe that should be back on the table and iterated on properly. other devices do it, and it doesn't seem to be an issue (eg. stop all sounds from playing by pressing EXIT five times on the sp404mk2.)

other alternatives:
1. tap shift (4-5) times within a determined window to enable 
2. tap shift 3 times and hold for a determined short window to toggle
3. tapping shift twice and holding to enable confirmation on screen, tap shift again to enable, any other pad to cancel
4. hold shift + tap learn/input twice
5. use the closest pad to shift: triplet view + shift, to enable. triplet view will have to be made effect on-release instead of on-press. this would be the best option for those with one hand and not really looking at their deluge.

if the feature has enough value to use just 1 highly utilized pad as a shortcut in it's current implementation, then there's good reason for it to be enabled by default for all users. now say if this feature was enabled by default but with _no_ shift lock light indication, it would be immensely confusing for users. people _would_ unintentionally toggle it on. There should be a safeguard for accidentally toggling this (more than one pad in shortcut to toggle, or additional taps/holds as suggested above) or else it'll just end up causing lots confused reports from users and ultimately be a disabled and unused feature that actually has value.

ok-reza, 2024-04-24:
> yeah the intention is for shift to not stick if another button is pressed before shift is released. It's possible that isn't working for some shortcuts, if you can find a pad which doesn't work that way we can fix it.

if i short press shift, press an audition pad and let go quickly, i'm able to toggle sticky shift.

dkrieger, 2024-04-24:
> > yeah the intention is for shift to not stick if another button is pressed before shift is released. It's possible that isn't working for some shortcuts, if you can find a pad which doesn't work that way we can fix it.
> 
> if i short press shift, press an audition pad (in order to audition silently) and let go quickly, i'm able to toggle sticky shift.

If this is indeed the case, I suspect resolving this would make most users tolerant of a longer window and the end result would be preferable to most

m-m-adams, 2024-04-24:
As far as I can tell that only happens if you release shift before pressing the audition pad. It selects silently because of sticky shift, otherwise it would make a noise

closed 2024-05-06 by m-m-adams