SynthstromAudible/DelugeFirmware #1730: Crash while loading songs

sichtbeton, 2024-04-15:
### Please describe the problem:

opening this song crashes the deluge with QR Code.

[SONG008 2.XML.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14982842/SONG008.2.XML.zip)

[image]

### What is the expected behavior?

open the song as usual.

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_04_14-7d90449

### If possible provide the steps to reproduce the issue and upload additional media:

open the song.

sapphire-arches, 2024-04-15:
E411 is "missing patched param".
```
arm-none-eabi-addr2line -Capife build/Release/deluge.elf 0x2011c6f5 0x2011c707 0x20100b65 0x200a7595 
```
This is probably due to the backup param manager stuff from #1658, I don't think I initialized the patched param manager.

sichtbeton, 2024-04-20:
basically all songs saved prior to this nightly cant be opened. they simply crash. stumped.

sapphire-arches, 2024-04-21:
This was indeed caused by #1658.

closed 2024-04-21 by m-m-adams

sichtbeton, 2024-04-27:
Thanks for fixing this!

commit df7a967b (2024-04-21): song: limit backup param manager insertion to audio clips (#1747)

other clip types already have recovery logic in other places, 

Fixes #1730
PR #1747 (2024-04-21): song: limit backup param manager insertion to audio clips
Other clip types already have recovery logic in other places.

Fixes #1730
PR #1752 (2024-04-21): song: limit backup param manager insertion to audio clips (#1747)
other clip types already have recovery logic in other places,

Fixes #1730
commit 2e40b98e (2024-04-21): song: limit backup param manager insertion to audio clips (#1747)

other clip types already have recovery logic in other places,

Fixes #1730