SynthstromAudible/DelugeFirmware #1788: Disappearing Keyboard Mode

redlunch, 2024-04-28:
### Please describe the problem:

The keyboard mode gets deactivated after switching from a kit track to a synth or midi track. The pads still register MIDI to a daw, but they do not light up to indicate what note you're playing.

### What is the expected behavior?

The pads light up to show you each octave or note within a scale

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

Deluge C1_0_1

### If possible provide the steps to reproduce the issue and upload additional media:

Create a new track with a synth, change it to keyboard mode. Then switch the track to a kit, then back to a synth or MIDI track, the lights indicating different notes will not be lit.

seangoodvibes, 2024-04-28:
Can you try the 1.1 beta please? I just tried your exact steps and I do have this issue.

Can you also post a video demonstrating your issue?

You can upload the video directly to here by dragging it into your comment.

Thank you

m-m-adams, 2024-04-28:
I believe this was reported and fixed a while back

closed 2024-04-28 by m-m-adams