SynthstromAudible/DelugeFirmware #1789: Getting "file not found" when pressing LOAD+SYNTH or turning SELECT knob

chewabledrapery, 2024-04-28:
### Please describe the problem:

After flashing c1.1.0 Beethoven when I try to load a preset or scoll to next i get "file not foun error"

### What is the expected behavior?

To bring u a browse list or scroll to next/previous preset

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

deluge-v1_1_0-beta+2024_04_28-cc25fff.bin

m-m-adams, 2024-04-28:
I'm not able to reproduce this on my deluge 

Would you be able to install a debug build? There's a site here https://bfredl.github.io/delugeclient/app.html that will connect to the deluge over midi and get better error messages

m-m-adams, 2024-04-28:
Actually I bet your issues are related - it might be failing to load the preset properly, while browsing it doesn't try again until you press select so you're getting the file not found error

chewabledrapery, 2024-04-28:
> Actually I bet your issues are related - it might be failing to load the preset properly, while browsing it doesn't try again until you press select so you're getting the file not found error

Yeah that's what I was thinking tbh, i've put an example project in #1790

Thanks!

chewabledrapery, 2024-04-28:
> I'm not able to reproduce this on my deluge
> 
> Would you be able to install a debug build? There's a site here https://bfredl.github.io/delugeclient/app.html that will connect to the deluge over midi and get better error messages
> 
> Here's where you can get a debug build - https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/8863312131

So I hook my deluge up to the USB of the PC then load this page? I'll give it a go... Thanks!

chewabledrapery, 2024-04-28:
> I'm not able to reproduce this on my deluge
>
> Would you be able to install a debug build? There's a site here https://bfredl.github.io/delugeclient/app.html that will connect to the deluge over midi [...]

OK so got the deluge hooked up, displaying debug info and emulated display.

I still have the arpeg issue but not the "file not found" issue on this debug build...

m-m-adams, 2024-04-28:
Alright, thanks for checking. Not sure what the issue would be then

m-m-adams, 2024-04-28:
Is this happening with songs started on community? If it's just from official then possibly a duplicate of #1792

ThePuny, 2024-04-28:
> Is this happening with songs started on community? If it's just from official then possibly a duplicate of #1792

I am having the same issue as described also for songs started on Community.

m-m-adams, 2024-04-28:
Yeah there's a PR to fix it, the bug is caused by the song having preset file paths that no longer exist on the card. With the bugfix linked above (#1794) it'll just restart from the root synths folder when that happens

closed 2024-04-28 by sapphire-arches

chewabledrapery, 2024-04-29:
> Is this happening with songs started on community? If it's just from official then possibly a duplicate of #1792

I definitely started this project on community, been on community since the first release.

chewabledrapery, 2024-04-29:
> Yeah there's a PR to fix it, the bug is caused by the song having preset file paths that no longer exist on the card. With the bugfix linked above (#1794) it'll just restart from the root synths folder when that happens

if that is the case I'll look through all my project xml and make sure all paths are valid...

PR #1794 (2024-04-28): Bugfix/recover from not finding synth on card
Fix #1792 
Fix #1789 

For projects where the synth is no longer in it's original location it will be loaded as "exists on card" but might not be in it's original location. In this case we'll cache that it's not on card but not return an error, as that's signalled through changing the existsOnCard attribute and the null file pointer.
PR #1798 (2024-04-28): Bugfix/recover from not finding synth on card
Fix #1792 
Fix #1789 

For projects where the synth is no longer in it's original location it will be loaded as "exists on card" but might not be in it's original location. In this case we'll cache that it's not on card but not return an error, as that's signalled through changing the existsOnCard attribute and the null file pointer.