SynthstromAudible/DelugeFirmware #1790: Arpegiator not there on presets when loading a project

chewabledrapery, 2024-04-28:
### Please describe the problem:

After flashing c1.1.0 Beethoven when I load a previous project any synth patch that had an arpegiator has lost its arepgiator. I'm not sure if this is related to the "file not found" issue I've just opened (#1789)

### What is the expected behavior?

The arpegiator settings should be recalled correctly

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

deluge-v1_1_0-beta+2024_04_28-cc25fff.bin

seangoodvibes, 2024-04-28:
Hi

Can you upload the song files / synth presets that had arp that you're noticing this problem with?

You can save your previously project with save collect all to share your whole project.

Thank you

chewabledrapery, 2024-04-28:
Example project attached, problem patch is "BD06 - 07 - PLUCK", arpeg works prior to latest beta, its doing this on other arpeg patches in other projects along with the "file not found" issue i've also logged...
[Breaks 13.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/15140801/Breaks.13.zip)

Hope this helps, thanks!

soymonitus, 2024-04-28:
Not a bug, this is due to not having added the ability to read the old arpeggiator "mode" parameter from the XMLs till very recently. Then loading an instrument that was created with probably one of the nightlies which didn't have that, and then the XML file version was set to 1.1.0 but the "mode" hadn't been imported and transformed to arpMode, octaveMode and noteMode.

After loading the song, setting the arp parameters again, and resaving it again, will fix the issue.

On the other hand while testing I found another different bug, reported here: https://github.com/SynthstromAudible/DelugeFirmware/issues/1797

m-m-adams, 2024-04-28:
can we handle that case in any way? Like even if it's 1.1 store the "mode" parameter and convert later if we haven't found other arp params

soymonitus, 2024-04-28:
> can we handle that case in any way? Like even if it's 1.1 store the "mode" parameter and convert later if we haven't found other arp params

the thing is that we then rely on the order of parameters (if they were written with a 1.1 firmware that didn't take "mode" into account, then that same firmware was writing the new arp params first before the old ones, so when you read the "mode" one, it will override the other new params already read. 
Or do you mean that even when reading the old "mode", before setting the new arpMode, octaveMode and noteMode, I check for their values and if I find something other than the default ones (off, up, up, respectively), I then do nothing??

soymonitus, 2024-04-28:
Something like this?
[image]

m-m-adams, 2024-04-28:
Yeah exactly - that'd be great

closed 2024-04-29 by m-m-adams