SynthstromAudible/DelugeFirmware #1792: "File not found" when adding tracks to projects started on Official firmware

MkUltra85, 2024-04-28:
### Please describe the problem:

Some (but not all) projects that were started with the official firmware will not allow me to create new synth / kit / midi tracks. When I press a new row in Song view (or when I press the row button in Arranger view), it gives me a "file not found" error. I can, however, still copy existing tracks. 

### What is the expected behavior?

Creating a new track.

### Is there a relevant Pull request?

I don't believe so

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

Community C1.1 Beethoven Beta

m-m-adams, 2024-04-28:
Can you share an example? I've got ideas for a fix but don't have a way to test it

m-m-adams, 2024-04-28:
As a temporary fix you can probably save and reload the song

MkUltra85, 2024-04-28:
[MK - Jul 11 3.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/15142426/MK.-.Jul.11.3.zip)

Thanks for the quick response. Here's a project it happens with. I tried saving / reloading, but it didn't fix.

m-m-adams, 2024-04-28:
Confirmed I can reproduce, I'll update when I have a fix

PR #1794 (2024-04-28): Bugfix/recover from not finding synth on card
Fix #1792 
Fix #1789 

For projects where the synth is no longer in it's original location it will be loaded as "exists on card" but might not be in it's original location. In this case we'll cache that it's not on card but not return an error, as that's signalled through changing the existsOnCard attribute and the null file pointer.
PR #1798 (2024-04-28): Bugfix/recover from not finding synth on card
Fix #1792 
Fix #1789 

For projects where the synth is no longer in it's original location it will be loaded as "exists on card" but might not be in it's original location. In this case we'll cache that it's not on card but not return an error, as that's signalled through changing the existsOnCard attribute and the null file pointer.
closed 2024-04-28 by sapphire-arches