SynthstromAudible/DelugeFirmware #1793: Text scrolling is too slow

sapphire-arches, 2024-04-28:
### Please describe the problem:

Text scrolling is noticeably slower/less smooth than previously.

### What is the expected behavior?

Scroll behavior should match pre task-handler.

### Is there a relevant Pull request?

#1708

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

m-m-adams, 2024-04-28:
Hm yeah - I wonder if the ui timer manager needs to be called more often

sapphire-arches, 2024-04-28:
I think we might be calling USB stuff too infrequently as well, I get checksum errors trying to use sysex FW loading now.

m-m-adams, 2024-04-28:
Ok, I think that one is in the playback handler task and probably shouldn't be

PR #1808 (2024-05-01): Increase ui and encoder timer rate
Fix #1793 
Fix #1796 

Just increase their frequency
closed 2024-05-01 by sapphire-arches