SynthstromAudible/DelugeFirmware #1795: Recorded notes in kits do not always play in playback

stefanherzog, 2024-04-28:
### Please describe the problem:

In a kit some of the recorded notes wouldn't play/sound even though the mute state of the kit row is green and the sample does play when playing the audition pad. When pressing the sample's mute state pad twice (i.e., change it to yellow and then to green again), then it does sound.

In case this matters, it was one of the classic drum machine kits. I haven't tried it with a new kit.

### What is the expected behavior?

All recorded notes in non-muted kit rows should play back.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.1.0-BETA-CC25FFF

### If possible provide the steps to reproduce the issue and upload additional media:

- create a new song
- Change current clip to kit
- Press play
- Press record and press the audition pads of several kit rows

Some of the kit rows do not play back even though their mute state button is green, indicating that it's not muted. But it seems quite unpredictable which kit rows are affected or how many.

For a non-sounding row, press the mute button twice (i.e., turn it yellow and then green again), and now it does sound.

sapphire-arches, 2024-05-05:
This appears to be related to quantization, assuming it's the same bug as ridemysailboat saw [on discord](https://discord.com/channels/608916579421257728/1138924035657449482/1236733990950670459).

It might also be related to #1819.

soymonitus, 2024-05-05:
I also experience this bug on every kit. Some of the rows (I couldn't define the pattern) get complete silent after recording some notes. They become unrecoverable. It doesn't matter if i clean the clip (<> +Back) if i input the notes manually they don't sound either. Quite frustrating, and super easy to reproduce

soymonitus, 2024-05-07:
With latest code from community branch I can confirm this issue is fixed!

closed 2024-05-07 by m-m-adams

soymonitus, 2024-05-07:
Not sure why but it worked with my songs, but now that I created a different song it doesn't work, it needs to be reopened @m-m-adams or if not, then consider this new bug report with more explanations of how to reproduce:
https://github.com/SynthstromAudible/DelugeFirmware/issues/1859

stefanherzog, 2024-05-14:
I'm still getting this on 1.1.0-BETA-A840144. Can someone try this with the AT RHYTHM kit? Maybe it only shows up with old kits now?