SynthstromAudible/DelugeFirmware #1807: Can not map parameters / note ranges to MIDI via USB or MIDI input

lilscribby, 2024-04-30:
### Please describe the problem:

Unable to map parameters or synth track's notes. Tested on a new synth track created when powering on the Deluge.
Tried connecting to:
* a MIDI keyboard connected via USB-A (Deluge is plugged in via AC adapter and MIDI keyboard is being powered via USB)
* a MIDI expression pad (Touche) connected via MIDI IN port.

### What is the expected behavior?

* For a synth track, LEARN+AUDITION+(press a note on the keyboard) should map the note range to the keys .
* For a parameter (while it is blinking/selected in the grid) LEARN+GRID+(input MIDI channel) should map the MIDI channel to that parameter.

This is the behavior of the official firmware v4.0.1.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus, Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_04_30-2126296 and deluge-c1_0_1

### If possible provide the steps to reproduce the issue and upload additional media:

1. Connect MIDI device via USB or MIDI-IN.
2. Power on Deluge.
3. Hold LEARN+AUDITION and press a key on the MIDI controller. The range is not mapped and the keys do not flash white as expected.
4. Enter a parameter, for example SHIFT+[FREQUENCY].
5. Hold LEARN+[FREQUENCY] and press a key on the MIDI controller. The parameter is not mapped to the device.

m-m-adams, 2024-04-30:
I can't reproduce - works fine on 1.0.1, 1.1, and 1.2. Could you upload a video please? How is the touche being powered?

Also scroll through the settings in midi->devices->your usb device->mpe and check that it all looks right.

lilscribby, 2024-05-01:
I don't understand what happened, I started it up with 1.0.1 and it works fine now. Didn't change any settings, controllers are plugged in the same as they were before. Ghosts in the machine... Guess there's no problem?

closed 2024-05-01 by lilscribby