SynthstromAudible/DelugeFirmware #1816: quantized stutter

entzmingerc, 2024-05-02:
### Please describe the problem:

With quantized stutter FX set to ON in the community features, when you set the time length to `8ths` or `4ths`, activate the stutter, then turn the playback speed down to its slowest value, there will be noise happening at the lowest value for these 2 time lengths. I subjectively think it sounds very similar to the sound when you turn the playback speed up to the fastest setting, but I'm not sure if that's relevant information or not. This doesn't seem to occur for other stutter lengths (tested 16, 32, 64 without issue). It also does not seem to occur for unquantized stutter of any length value. 

### What is the expected behavior?

The expected behavior is the slowest playback speed setting where it sounds like it's lower pitch and sample rate of the buffer is a lot slower, similar to what happens near the minimum value. It seems like it's only the minimum value that's affected for the 8th and 4ths, but everywhere else in the stutter range seems fine. 

### Is there a relevant Pull request?

uhhhh beta! https://github.com/SynthstromAudible/DelugeFirmware/releases/tag/beta

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

other

### What is the firmware name:

deluge-v1_1_0-dev+2024_04_28-36063ed.bin

### If possible provide the steps to reproduce the issue and upload additional media:

1) Turn quantized stutter (STUT) set to ON in community features
2) Turn the gold knob for stutter to either 8ths or 4ths time length
3) Engage stutter by pressing down on the gold knob encoder
4) Hold down and turn the gold knob encoder all the way to the left until the playback speed is at the minimum value, you should hear a sharp difference between the minimum and second-to-minimum value.

5) (optional) instead of turning all the way to the left, turn it all the way to the right and let me know if the maximum value sounds like the minimum value, idk if this is relevant lol

soymonitus, 2024-05-03:
I am pretty sure that because now with the quantized stutter we access a wider range of speeds, starting the 4ths at the center of the parameter, we reach to a point where the integer jumps from negative values to positive values (overflowing I mean), or something like that. Maybe?

m-m-adams, 2024-05-03:
That'd be my guess as well

closed 2024-05-07 by sapphire-arches