SynthstromAudible/DelugeFirmware #1817: playhead glitch with shortened kit rows with midi loopback

entzmingerc, 2024-05-02:
### Please describe the problem:

Reproduced on nightly (1.2.0) and beta (1.1.0)
video 1: https://photos.app.goo.gl/YUWYiAt7ihzmi7UN8
video 2: https://photos.app.goo.gl/tom2xaLBNktRr6Lb6

There are a few errors. 

With midi loopback turned ON, you can use the new feature `Midi Learn to Kit` to learn an incoming midi sequence to a kit clip from song view. This will trigger kit rows with midi notes. However, if you shorten the length of a kit row, the incoming midi notes seem to jump the playhead of the shortened kit row back by a specific length of time. I tested this with a midi clip with one row full of 16th notes learned to an empty kit clip without notes to see the behaviors. 

- This glitch seems to occur only with the initial clip because when I create a new kit clip and learn it, the playhead glitching does not occur.
- If the initial kit clip was learned, then I delete the initial kit clip, then make a new kit clip that initializes to preset kit 0, then press [learn] in song view, it erroneously highlights the new kit clip as learned even though it was just recently created. 
- Any incoming midi note will glitch the playheads of all shortened kit rows.
- When the kit loops, the glitch seems to not affect the playheads for the first 16th note, but starts to glitch the playheads by the second 16th note coming in from the midi clip. 
- With a row of midi 16th notes, when the playheads glitch, you can see the kit playheads jump back in time the length of a 16th note. I tested with midi 32nd notes as well, and the kit lengths jumped back in time only 32th note length of time instead. 
- This seems tempo independent, you can slow it down to 30 BPM to observe easier. 
- Zooming in, the playheads seem to progress in sync with the tempo, its position seems to get pushed back in time when a midi note comes in. You can see the playheads scrolling and just jumping back in time. 
- Repeatedly pressing the kit row audition pad while a steady stream of midi 16th notes are glitching the playheads seems to help the playheads ignore the glitch and keep in time. 
- Placing notes inside the kit at every 16th note at the same time as notes from the midi clip will ignore the glitch from happening, it seems to only glitch a playhead when there aren't notes in that kit row. 

I think that's everything lol please check out the linked videos :)

### What is the expected behavior?

The expected behavior is to trigger the kit row when the kit receives a midi note as described by the `Midi Learn to Kit` feature. The playhead of the kit shouldn't be affected.

### Is there a relevant Pull request?

https://github.com/SynthstromAudible/DelugeFirmware/pull/1251

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly, other

### What is the firmware name:

deluge-v1_1_0-dev+2024_04_28-36063ed

### If possible provide the steps to reproduce the issue and upload additional media:

1) Turn deluge on, and make the initial clip a Kit by pressing Kit.
2) Make a new midi clip (Default to channel 1 I guess), put in 16 16th notes into a row to fill a row. 
3) Turn on `Midi Loopback (M_LP)` in song view menu
4) Press play to begin sending midi notes
5) [Learn] + row pad of the kit clip to learn the midi clip to the kit clip, you should hear the kick drum (assuming this is factory preset kit 0)
6) Inside the kit clip, hold an audition pad and turn the <> encoder to the left to shorten a kit row. Observe the kit row playhead jumping back in time.
7) Turn down the tempo to a slower BPM (idk 30?) to observe the patterns more easily.

Delete the kit clip, make a new kit clip (which should initialize to preset 0 again), then press learn without pressing anything else and you should see that the new kit clip says it's learned when it hasn't been learned yet. 

It seems that the first note in the clip lets all playheads progress normally. Then for the next note (second 16th note), if any shortened kit row is at the last 16th note square, it will proceed all shortened kit rows normally again and loop that playhead position. Otherwise, it will glitch all playheads of all shortened kit rows.

sapphire-arches, 2024-05-14:
It looks like we're re-processing ticks a bunch, if the secondary row is set to 1/2 a bar while the kit is set to 1 bar it advances once per bar.

sapphire-arches, 2024-05-15:
It also doesn't reproduce if fewer notes are sent out of the midi channel. My suspicion is that something in this loop is poking the glitching note row too often: https://github.com/SynthstromAudible/DelugeFirmware/blob/f636f84f7cd938423d912bedc54f82016b7e8d3b/src/deluge/model/clip/instrument_clip.cpp#L721-L805

m-m-adams, 2024-05-23:
Pulled loopback out of 1.1, moving this to next release