SynthstromAudible/DelugeFirmware #1818: VU Meter rendering is deleted by pad press and release in performance view

entzmingerc, 2024-05-02:
### Please describe the problem:

https://github.com/SynthstromAudible/DelugeFirmware/pull/1812
The link to the PR above fixes a problem where vertical scrolling would make the VU meter disappear in performance view.

After playing audio, then turning play off and looking at the VU meter as the sound decays to silence, the VU meter seems to show 2 rows lit up at the bottom with no sound being output. This sometimes goes away if you do an action that retriggers rendering of the sidebar like vertical scrolling, but I'm not sure I can reproduce this consistently. 

In performance view, when I press a pad or release a pad it kills the render of VU meter. Tap the pads fast to reproduce the rendering glitch. Sometimes when holding a stutter pad it doesn't render the VU meter, with quantized stutter at 32nd and 64th note rates it deletes the VU meter render a bit more frequently.

### What is the expected behavior?

When pressing and releasing a pad in performance mode, the VU meter rendering should remain unaffected and not disappear from the pads.

If there is no audio being output on the main outputs, then the VU meter should show no illuminated pads. 

### Is there a relevant Pull request?

possibly this one https://github.com/SynthstromAudible/DelugeFirmware/pull/1369

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

other

### What is the firmware name:

deluge-v1_1_0-dev+2024_04_28-36063ed

### If possible provide the steps to reproduce the issue and upload additional media:

1) Turn on VU Meter by following instructions here <https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/community_features.md#418---added-vu-meter-rendering-to-song--arranger-view>
2) Play a song to get some audio running through the VU Meter and switch to performance view from song view by pressing the keyboard button
3) Quickly tap and release pads while in performance view and observe the VU Meter rendering sometimes being deleted.
4) Try tapping fast stutter speeds and holding them to catch the VU Meter rendering being deleted until a pad is released. 

Turn playback off and watch VU Meter decay towards zero, I think it should disappear and lights should turn off but sometimes it didn't, if you can't reproduce feel free to ignore this part and I'll try finding a reproducible pattern.

seangoodvibes, 2024-05-03:
Confirmed, also confirmed this happens in arranger automation view

seangoodvibes, 2024-05-06:
@entzmingerc Hey Cameron, can you confirm whether this has been fixed in nightly and beta and close if it has? Thanks!

entzmingerc, 2024-05-07:
Doesn't seem fixed on nightly. deluge-v1_2_0-nightly+2024_05_07-5e90233
Seems to have less flickering happening on beta but still is flickering while tapping some buttons. deluge-v1_1_0-beta+2024_05_07-744d133

instead of re-rendering could you cheat and just redisplay the most recent sidebar render instead of trying to rerender all the computation to smooth it out for each button press? Just an idea idk

closed 2024-05-15 by sapphire-arches