SynthstromAudible/DelugeFirmware #1819: beta: no midi playback from one sequencer row after recording

nikodemus, 2024-05-03:
### Please describe the problem:

After recording a 1 bar sequence using the audition pads to a MIDI clip, one of the rows did not send MIDI out (verified using a midi monitor). Other rows worked normally. The faulty row's audition pad worked normally. Saving the song and reloading made the row work normally.

Creating a new song and attempting to reproduce did not result in the same behaviour. I do not recall if I had done other things in the same song before this occurred. I _think_ it was a recently initialized fresh song, but I am not 100% certain.

In addition to the MIDI clip there was just one audio clip monitoring the external instrument.

If this re-occurs I will record video to demonstrate, but if there's anything else that would help debugging, please let me know?

### What is the expected behavior?

MIDI should play from all sequencer rows.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.1.0-beta-95acd60

m-m-adams, 2024-05-03:
Hm, I wonder if there's an unitialized value somewhere in the clip constructor

soymonitus, 2024-05-05:
Definitely an issue with Quantization, when I set it to Off, the bug is gone

PR #1850 (2024-05-06): consider notes at end of sequence after wrapping
Fix #1819
PR #1853 (2024-05-06): consider notes at end of sequence after wrapping
Fix #1819
closed 2024-05-06 by m-m-adams