SynthstromAudible/DelugeFirmware #1829: Norns mode issues with Buoys project

creativecontrol, 2024-05-04:
### Please describe the problem:

The norns grid project Buoys https://norns.community/buoys has some unexpected behaviors on Deluge. Pads 2 & 3 on the bottom row never light up and Record mode is constantly triggered causing the Record button to light up and flicker and a single red pad to travel across the screen. It also seems like being in record mode sometimes removes placed pylons (short press on a pad to place pylon). This can been seen in the video.

All other functions are working as expected (i.e. the MIDI commands to norns are working).
I have looked through the settings and made sure the Record function is not assigned to a MIDI control.

[image]
[image]

### What is the expected behavior?

Pads 2&3 should light during the two lit modes of the norns Buoys project (tides, buoy).

Record mode should not be activated.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

other

### What is the firmware name:

C1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on Deluge
2. Plug USB into norns (I'm using the shieldXL DIY with RPi4 and firmware 220306)
3. Press SHIFT+Settings and scroll the FEAt
4. Press Select
5. Scroll to norn
6. Press Select
7. Scroll to on
8. Set current Clip to MIDI and scroll to channel 16
9. Press and hold Keys and Scroll Select to norn

entzmingerc, 2024-05-04:
Hmm. I'll give this a whirl this afternoon.

entzmingerc, 2024-05-04:
@creativecontrol 
I had no idea about buoys, really beautiful script

Beta 1.1.0 from 4/28 seems to work fine with buoys
c1.0.1 seems to work fine as well on my end. There are a few hiccups where some LEDs are erroneously being lit up when the tide passes through, but I'm unable to recreate the pad 2&3 LEDs from not lighting up. 

It looks like you already checked that REC isn't learned to a midi command. I can get erratic REC flashing when the midi command is learned to the incoming midi notes that are sent from the midi clip when you press a pad. I'm not getting immediate glitching behavior when loading buoys or reloading the script. I'm not sure what midi command would trigger REC to turn on, but if you go to MIDI > CMD > REC it should be none, also in the same menu PLAY should be none, or both should be learned to a different midi channel before loading buoys. 

Two other strange behaviors I noticed in your clip is at like 5 seconds in the LEDs by (synth, kit, midi, CV) those shut off, and then the recording playhead turns red as well as it scrolls across the top row. I'm able to reproduce this if I go to MIDI > CMD > LAYER and press and hold [learn]. This will learn the midi data coming in from midigrid on channel 16 to the layer command I think and I get erratic REC flashing and the playhead turning red. I'd make sure to check that all these midi commands are set to none by scrolling to them and turning the select knob to the left until it says none.

creativecontrol, 2024-05-05:
Thanks so much for testing. It turns out I had all of the MIDI > CMD options set to none except for Layer. I somehow missed that every time I checked those assignments. I still seem to get a random unlit pad every once in a while but that is much better than the scrolling red dot.

Thanks again.

entzmingerc, 2024-05-05:
Great! Random unlit pad every once in a while is expected. I think it's because of midigrid, but it could be anything lol.

sapphire-arches, 2024-05-05:
seems like this is resolved, so I'll close this issue? Bouys seems to send a huge number of updates to the grid, we'd probably need to optimize it on the midigrid side to reduce the probability of pads going dark.

closed 2024-05-05 by sapphire-arches