SynthstromAudible/DelugeFirmware #1864: rotating through different scales is now destructive

nikodemus, 2024-05-08:
### Please describe the problem:

With scales containing different numbers of notes in the rotation, going through the whole cycle is destructive.

Suggested solution:

1. Require activating scales with other than 7 notes from the community menu.
2. If there are non-root notes in the current _song_, skip remove scales with different number of notes than the current one from the rotation. Ie. if I have picked a pentatonic scale and started a song, the scale button should rotate only between pentatonic scales, without showing the diatonic ones.
3. Add a Scale-menu, accessible with SCALE + SELECT, from which any scale can be selected, even those with different number of notes. If there are non-root notes in the current song, and the number of notes is different in the new scale, ask for confirmation.

(As a bonus, the scale menu would make it easy to eg. switch back and forth between major and minor without having to rotate through several scales every time.)

### What is the expected behavior?

Going to a different scale and coming back to the original should restore the notes.

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.1.0-beta-95acd60

### If possible provide the steps to reproduce the issue and upload additional media:

1. Start with any diatonic scale, and a 7th chord. Say MAJOR 1, 3, 5, 7.
2. Press SHIFT + SCALE to cycle through different scales, until you're back in MAJOR.
3. Observe that your chord is now 1, 3, 4, 5 instead.

soymonitus, 2024-05-08:
I would vote for a variation of option 3, which is:
- Override the shortcut Shift + Scale to show a popup like the one in FILL/ONCE clips with the list of scales (instead of just cycling), confirm selection with Select click. If the target scale has, for example, 5 notes but your sequence contains more than 5 rows with notes, it will print CANT, being consistent with the Keyboard Sidebar scale view.

soymonitus, 2024-05-08:
I will try to solve this by making the deluge  "remember" the notes placement in case you don't edit the sequence and just cycle all scales back to same scale

nikodemus, 2024-05-09:
So, I took a look at making the low effort "root notes only" patch for my own use if nothing else, but it doesn't really work:

- New song
- Pentatonic minor
- Write some notes
- Cycle through scales... oops, you can't return to pentatonic anymore, using the "only root notes" logic.

The "shared scale degrees only" logic would probably work, but it might not be super musical. (If @soymonitus gets the memoization done, that's great, but I've already broken a couple of songs with the cycling, so I'm keen to fix this for myself.)

nikodemus, 2024-05-09:
On further reflection, I do think the caching is the right way. Other options are just as complex, and still more constrained. Below some notes. I assume @soymonitus has already thought about this more than I have, but in case it's useful, here's the way I would attack this:

Keep two maps: 7 <-> 6 and 6 <-> 5. The 7 <-> 5 map is implicit: to go from 5 to 7 you go first to 6 and then to 7.

A map identifies the elided scale degrees when going from a larger scale to a smaller one. In the current setup this means a map can be represented by a single integer.

Example: 7-6 map = 2, 6-5 map = 3, and a transition from 5 note scale to 7 note scale.

First to 6-note scale:

1->1
2->2
X->3 no translation!
3->4
4->5
5->6

To 7-note scale:

1->1
X->2 no translation!
2->3
3->4
4->5
5->6
6->7

Ie. the combined effect of reversing these maps was: 1->1, 2->3, 3->5, 4->6, 5->7

When you come back to 7 note scales, you can drop the maps, since they will be regenerated consistently when going to smaller scales, assuming notes aren't added to degrees that were previously empty.

If there are less notes in the larger scale than the smaller scale can accommodate, the mapping
can be picked freely of any of the unused scale degrees in the larger scale. There are probably more musical ways, but technically even something dead simple like "highest unused scale degree" will work. 

If there's ever a direct jump from 7 to 5 notes, it should be treated as if it was done in two steps, through a 6-note scale.

I hope this isn't too incoherent, it's getting a bit late :)

closed 2024-05-12 by sapphire-arches