SynthstromAudible/DelugeFirmware #1872: dbt toolchain v13 for windows has cmake 26.5, running dbt test requires cmake > 27

nikodemus, 2024-05-09:
### Please describe the problem:

Running "./dbt.cmd test" on a clean tree on Windows fails due to cmake version mismatch.

### What is the expected behavior?

Running dbt test should work.

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

commit 744d1332dad16da37a50e721d3a0329059ba95a4

### If possible provide the steps to reproduce the issue and upload additional media:

Running "./dbt.cmd test" on a clean tree, on release/1.1 branch:

```
$ ./dbt.cmd test
C:\Users\nikodemus\src\DelugeFirmware2\toolchain\v13\win32-x86_64 missing, will install...
C:\Users\nikodemus\src\DelugeFirmware2\toolchain\v13\win32-x86_64\VERSION missing, will install...
DBT: starting toolchain upgrade process from 0 to 13
Downloading Windows toolchain..done!

    Directory: C:\Users\nikodemus\src\DelugeFirmware2\toolchain

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----          5/9/2024  12:01 PM                v13
Extracting Windows toolchain..moving..done!
Cleaning up temporary files..done!

C:\Users\nikodemus\src\DelugeFirmware2
-- The C compiler identification is Clang 16.0.0 with GNU-like command-line
-- The CXX compiler identification is Clang 16.0.0 with GNU-like command-line
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/LLVM/bin/clang.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/LLVM/bin/clang++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at C:/Users/nikodemus/src/DelugeFirmware2/build/tests/_deps/cppspec-src/CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.27 or higher is required.  You are running version 3.26.5

-- Configuring incomplete, errors occurred!

$ toolchain/v13/win32-x86_64/cmake/bin/cmake.exe --version
cmake version 3.26.5

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

nikodemus, 2024-05-09:
The 3.27 requirement comes from cppspec-src, it seems.

sapphire-arches, 2024-05-09:
Please try `./dbt nuke` and deleting `toolchain/v13`. I'm not sure why but it seems like it's not picking up the fact that the latest and greatest toolchain (which should have the newer cmake) is v15.

nikodemus, 2024-05-09:
release branch for whatever reason is sticking with 13, but community branch for 15.

closed 2024-05-09 by nikodemus