SynthstromAudible/DelugeFirmware #1886: running spec tests on windows segfaults

nikodemus, 2024-05-10:
### Please describe the problem:

After building tests with Clang 16 using ./dbt.cmd test, running /build/tests/spec/Debug/all_specs.exe segfaults after test selection.

Backtrace from WinDbg:
```
[0x0]   ucrtbased!strlen+0x31   0x868fd8f198   0x7ff6dc0868a3   
[0x1]   all_specs!std::_Narrow_char_traits<char,int>::length+0x13   0x868fd8f1a0   0x7ff6dc09121a   
[0x2]   all_specs!std::basic_string_view<char,std::char_traits<char> >::basic_string_view+0x2a   0x868fd8f1d0   0x7ff6dc083e34   
[0x3]   all_specs!CppSpec::parse+0x54   0x868fd8f210   0x7ff6dc081be3   
[0x4]   all_specs!semver_spec+0x33   0x868fd8f6b0   0x7ff6dc08158c   
[0x5]   all_specs!main+0x58c   0x868fd8f770   0x7ff6dc0d1de9   
[0x6]   all_specs!invoke_main+0x39   0x868fd8f810   0x7ff6dc0d1f1e   
[0x7]   all_specs!__scrt_common_main_seh+0x12e   0x868fd8f860   0x7ff6dc0d1f9e   
[0x8]   all_specs!__scrt_common_main+0xe   0x868fd8f8d0   0x7ff6dc0d1fbe   
[0x9]   all_specs!mainCRTStartup+0xe   0x868fd8f900   0x7ff96bcb257d   
[0xa]   KERNEL32!BaseThreadInitThunk+0x1d   0x868fd8f930   0x7ff96c58aa48   
[0xb]   ntdll!RtlUserThreadStart+0x28   0x868fd8f960   0x0   
```

### What is the expected behavior?

Not segfaulting :)

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

commit c9d98050428f3e55803840709a9b5591707a37e7

### If possible provide the steps to reproduce the issue and upload additional media:

```
$ ./build/tests/spec/Debug/all_specs.exe
Available tests:
  0. semver_spec
To run a test, enter the test number: 0
Segmentation fault
```

sapphire-arches, 2024-05-10:
This might be a bug in upstream, cc @stellar-aria

nikodemus, 2024-05-12:
The issue appars to be in the interactive test running on windows, when run as:

```
./build/tests/spec/Debug/all_specs semver_spec --verbose
```

it works fine:

```
SemVer
  parses a simple semver
  parses a semver with a pre-release tag
  with an alpha pre-release version number
    is less than its full release
    is less than its beta release
    is greater than the previous core version
```

stellar-aria, 2024-05-13:
Yeah I think this might actually be an upstream CTest bug, since I've run into it before

nikodemus, 2024-05-13:
Close?

closed 2024-07-21 by m-m-adams