SynthstromAudible/DelugeFirmware #1894: Audio looping is buggy when Deluge has just booted (it doesnt start on time)

soymonitus, 2024-05-11:
### Please describe the problem:

When doing audio looping just from a cold boot state, the audio recording doesnt start on the first beat

### What is the expected behavior?

Everything is synced

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### If possible provide the steps to reproduce the issue and upload additional media:

When you boot the Deluge
create a song from scratch
add empty clip
convert to Audio clip
hit play, enable metronome\
hold record and tap a pad in the row below, that will start recording audio
but it fails to start on time

soymonitus, 2024-05-11:
https://github.com/SynthstromAudible/DelugeFirmware/assets/6472762/56069b7a-633b-4c63-aaba-a586f7a7e459

soymonitus, 2024-05-11:
Tested several commits till i found that the issue started happening after this commit:
https://github.com/SynthstromAudible/DelugeFirmware/commit/36229e8bcecb5555bbf2c902a411f6c3d1992388
Reverting it fixes the issue

soymonitus, 2024-05-12:
hmmm don't you think this should go into 1.1? A user that boots the deluge and want to do live looping of audio would get an awful experience. Not all users will play samples in the Deluge, they might just be guitarist, or have a bunch of hardware synths and they just sequence them and record audio clips stems into the deluge. Whenever they record the first clip, it will leave screw the sequencer and have it stop all sound for a couple or seconds @sapphire-arches

sapphire-arches, 2024-05-12:
(posting here for posterity as well) we don't need this in 1.1 since 1.1 is still using the old FatFS and scheduling logic.

m-m-adams, 2024-07-21:
This should have been fixed by the PR adding yields and removing the sd routine. I can't reproduce it anymore

closed 2024-07-21 by m-m-adams