SynthstromAudible/DelugeFirmware #1905: UI glitch in arranger view when extending audio clips

pmatila, 2024-05-12:
### Please describe the problem:

When extending an audio clip in the arranger view, some pads in the above rows may get unlit.

### What is the expected behavior?

Extending a clip obviously should not affect other rows pads.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community, other

### What is the firmware name:

Reproducable on 1.1 and community as of May 12th 2024

### If possible provide the steps to reproduce the issue and upload additional media:

Steps to reproduce:

- create new project
- go to arranger view
- create two tracks with synth clips (can be empty) that fill the screen
- create third track as audio track, record less than full screen (say, four pads worth), stop
- lengthen the audio clip to (full screen length)
-> some of the pads in the synth clips above go unlit

Other observations
- Seems to be entirely cosmetic
- The unlit pads follow a pattern, every third (1, 4, 7, 10, 13, 16) pads go unlit depending on how much the audio clip is extended, and first on the row above and then the corruption extends to the earlier track too
- Does *not* happen on 1.0.1, but reproducable on both 1.1 and community branch as of May 12th

I can bisect if somebody has an idea for a reasonable range.

sapphire-arches, 2024-05-12:
You'd probably have to go all the way back to January or so and that bisect could get a little annoying. I suspect it's one of a few refactorings that's gone in to the pad rendering code between then and now...

pmatila, 2024-05-13:
Okay it was indeed a little annoying :sweat_smile: , but found it:
commit 1f914f157556edb148e9bec6e6322999ebca49bb (PR #406) broke arranger view to a point that it's not possible to test, commit 85997821be867768c0f9b823a70f878a029fddc7 is the first commit with current, described behavior.

sapphire-arches, 2024-05-13:
Thanks for bisecting, I suspect there some additional bug in the arranger view pad rendering code then.

closed 2024-05-15 by seangoodvibes

pmatila, 2024-05-15:
Thanks a lot for the quick fix! Seeing what's basically memory corruption is always a bit nerve-wracking :sweat_smile: