SynthstromAudible/DelugeFirmware #1907: Strange UI dead-end when moving audio clip from arranger to grid mode

pmatila, 2024-05-12:
### Please describe the problem:

If song mode is in grid-mode, moving an audio clip recorded in arranger to the song mode (hold clip, press song) and then pressing Song again to go back to arranger causes "Impossible from grid" error. In this state, most buttons do nothing, some give this error message but then hitting a launch/preview pad (even unused ones) seem to clear it, and then eg Song to arranger works again.

### What is the expected behavior?

Pressing song should take you back to arranger from grid.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community, other

### What is the firmware name:

Reproduced with 1.1 and community as of May 12th

### If possible provide the steps to reproduce the issue and upload additional media:

- create new project
- go to song mode, switch to grid mode
- go to arranger view
- create an audio track, record something
- hold the clip, click song (to bring it to song/grid view)
- click song mode to go back to arranger
-> "impossible from grid" error

Many/most buttons are simply unresponsive in this state. Hitting an empty launch or green/blue mode button "fixes" it, then going to arranger works.

 Happens also in 1.0.1 and nightly (May 12nd local) too, not possible in official

pmatila, 2024-05-13:
Poking a bit more at this, the "impossible" case prevention seems to work as intended in the other direction, ie trying to open a clip from grid into arranger. It's the other direction that goes into this limbo state, although the clip does end up in the grid and things seem to work, sort of.

If the intent is to prevent this action in the first place, I guess ArrangerView::moveClipToSession() should have a similar check. But then, moving things from arranger to grid *seem* to work, the UI glitch aside. Don't understand the surrounding code sufficiently to tell whether it's really safe or not...

sapphire-arches, 2024-07-27:
As mentioned in #2287, this also applies to non-audio clips.

PR #2395 (2024-08-07): Bugfix/moving clip to grid
This left you stuck in the UI mode for holding a clip in rows mode since in the grid the clip you moved is unlikely to be under your finger

If switching to grid just set the UI mode to none instead

Fix #1907
PR #2397 (2024-08-07): Bugfix/moving clip to grid
This left you stuck in the UI mode for holding a clip in rows mode since in the grid the clip you moved is unlikely to be under your finger

If switching to grid just set the UI mode to none instead

Fix #1907
closed 2024-08-07 by seangoodvibes