SynthstromAudible/DelugeFirmware #1909: It should be possible to launch and enter a clip at the same time in grid view

sapphire-arches, 2024-05-12:
### Please describe the problem:

In grid view it's very common to want to copy a clip and then edit it. Currently this requires 6 button presses:
- Switch to blue mode
- Copy the clip
- Switch to green mode
- Launch the new clip
- Switch to blue mode
- Edit the new clip

This is irritating and can break the flow.

### What is the expected behavior?

A new latch state should be added to grid view equivalent to launch+edit so the workflow above becomes:

- In blue mode, copy the clip
- Press green+blue to enter "clip launch + edit" state. The blue and green mode buttons should blink (or both be lit) in this state.
- Press the copied clip to both launch and enter it.

This mode would latch like the other display modes.

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

other

sdesign75, 2024-05-17:
Would also like the following behaviour added if possible…

In grid view, when Default Active Mode is set to SELECTION, could we make the green and blue pads behave like the Shift button? So if you press and hold it goes into the Blue mode and when you release it then it returns to Green mode and vice versa. If you tap it latches.

Exmpl-xe, 2024-08-06:
A similar idea excluding both green & blue modes as being separate. Lite hold clip to enter, hold + new to copy & tape to launch

closed 2024-12-28 by m-m-adams