SynthstromAudible/DelugeFirmware #1920: Kit with samples longer than 2sec, mode Once, when audition oK but when recording they get super short

soymonitus, 2024-05-13:
### Please describe the problem:

When I create a new kit made of samples that are longer than 2 seconds, then I have to do the shorcut Mode, hold Affect entire, then change all samples to Once. They are now playing fully when hitting the audition pads. But when I start playing **and/or** recording, and then use the audition pads, the hits sound like in Cut mode, and if the sound is long enough and I hold an audition pad, the moment the playhead reaches the end of the clip, it cuts the sound.
When I record, the notes have the correct length (16th note), but they sound like if they were a 64th note in Cut mode.

Firmware 1.1 beta

### What is the expected behavior?

Sounds play fully in Once mode when Play is engaged and are not cut when playhead reaches the end.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

soymonitus, 2024-05-13:
https://github.com/SynthstromAudible/DelugeFirmware/assets/6472762/368a7b0b-302c-4b0e-8cfb-daaee42576c4

soymonitus, 2024-05-13:
Some clues if you see the video:
1 - After pressing play and using the audition pads, there is a moment they play fine, but if a sample was playing and playhead reaches the end, then that kit row because a super short sample time
2 - If you press an audition pad and the sound reaches the moment in time you pressed it before, it acts the same as if the playhead had reached the end
So it seems like if there were some variable saved per row that once it is set to certain value, the row doesn't play fully anymore.
3 - This happens with the beta from 2 weeks ago and with the beta from today

soymonitus, 2024-05-13:
Ohhhh I think I know why this happens, I did a LOAD ALL action in the kit creation and the folder had like 100 long tail drum samples. After using the Clean Unused and leaving just 16 samples or so, they play fine. Anyway... I am curious to know, which is the technical explanation that samples do that weird behavior? Memory getting full?

closed 2024-05-13 by soymonitus

ok-reza, 2024-05-13:
i can't remember exactly, but this behavior was introduced in either 4.0, 4.1.4 (post OLED firmware), or the first community firmware. i used to be able to load 75+ drum samples and not have the cut off issue you described during playback. worth eventually looking into to see if anything specific is the culprit or if any optimization is possible