SynthstromAudible/DelugeFirmware #1931: High CPU Indicator option not available

Akeboshiwind, 2024-05-14:
### Please describe the problem:

To replicate:
1. Turn on
2. `SHIFT` + `SELECT`
3. `DEFAULTS`
4. No `HIGH CPU INDICATOR` option

### What is the expected behavior?

A `HIGH CPU INDICATOR` option to be available

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

1.1.0-BETA-3BF0444

seangoodvibes, 2024-05-15:
It should be in tonight's beta. It was added today as it somehow got left out by mistake.

seangoodvibes, 2024-05-15:
> It should be in tonight's beta. It was added today as it somehow got left out by mistake.

Try this!
https://github.com/SynthstromAudible/DelugeFirmware/releases/download/beta/beta.zip

closed 2024-05-15 by seangoodvibes

Akeboshiwind, 2024-05-15:
Amazing thanks!