SynthstromAudible/DelugeFirmware #1938: HPF ladder filter FM (morph) cuts out sound around 6-8 on initial synth

entzmingerc, 2024-05-15:
### Please describe the problem:

Turn on Deluge, init synth, hold an audition pad to hear the synth, turn up HPF morph to 6 - 8 and it sorta distorts then cuts out audio then comes back. If you turn up the resonance, than you can hear the HPF stuttering a bit more noticeably around morph 4-9. If HPF cutoff is not set to 3 or greater then sweeping the morph seems to not cause the issue to appear. 

Setting morph to like 2 then increasing LPF resonance causes it to drop in volume almost immediately. 
Setting OSC1 TYPE to triangle seems to not be as noisy or messed up?

Ignorable subjective qualitative note: filter FM amount doesn't seem to affect the sound until I get it to self-oscillate with resonance around 35 ish, then when I sweep the morph I can hear some difference in the sound, but that seems like an extreme case for user. Is there some way to amplify FM-ing of the HPF cutoff such that it FMs it "more"? Idea would be out of scope for beta 1.1.0 tho.

### What is the expected behavior?

Sound should not cut out while turning up the HP_L morph filter FM amount between values 6-8 while HPF cutoff is at OFF for the initial synth. 

(otherwise I'm not sure what to expect when turning up HPF FM sometimes lol)

### Is there a relevant Pull request?

~

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

other

### What is the firmware name:

deluge-v1_1_0-beta+2024_05_14-3bf0444

### If possible provide the steps to reproduce the issue and upload additional media:

Turn on Deluge, init synth, hold an audition pad to hear the synth, turn up HPF morph to 6 - 8

Adjust LPF resonance while HPF morph is at 2 to hear it bounce around.

Set OSC1 TYPE to triangle to hear it not be that bad for some unknown reason.

m-m-adams, 2024-05-21:
It's hugely dependent on what's going into it, and currently doesn't do negative cutoff frequencies. Dunno if it's really a bug in either case though - it's pretty similar to FMing my analog filters

m-m-adams, 2024-07-21:
Looked again and I don't think it's a bug

What's happening is the square wave modulation of the cutoff at that depth results in perfectly out of phase output whenever it switches. Changing the filter routing to be high to low or using any other wave shape fixes the problem.

closed 2024-07-21 by m-m-adams